MartinNohr/MagicImageWand #1: progress and suggestions

MartinNohr, 2020-12-15:
Let's continue our discussons from the loghtwand post here.
I'll start by saying that I now have my test code on the TTGO getting closer.
- SD card works, even with exFat formatted cards larger than 32GB.
- Rotary dial switch works
- LED strip works
I have some text working on the display, but I have some work left to do to get the full menu system up and running. Later this week if I don't get too many interruptions.
I also have a schematic that I'll post tomorrow.
I'll have the code up as soon as it works.

DaveStu, 2020-12-19:
Hi gentlemen, hope all are well. It's been a while, been busy with covid life and the ever changing rules.
Anyway, seems much has been progrssed, the excitement is overwhelming.
Haven't had much time to design enclosures and attaching mechanisms, though I did look at Dirks design on previous thread and looked great, hows that going?
Whilst I've been away/waiting, ordered another LED strip, the temptation for a £5 deal with coupons was too good pass.

So, are we still using Arduino to compile?

MartinNohr, 2020-12-20:
Sounds fun. :-(
It can be compiled and it is running more or less on the TTGO now. I have a lot of display issues to resolve, but most of them should be done in the next couple of days. The LED's are running ok as far as I can tell.
I'm going to improve the fonts a bit, I think they are a little bit too small. I can read them ok, but I have really good eyes for a 70+ old engineer. I know a lot of my friends won't be able to read them.
As soon as I get that stuff done I'll add the preview feature and then start on other ideas that we think (or thought) of.

MartinNohr, 2020-12-20:
We've had a number of our friends get Covid now, One died, he was 5 years older than me. Kind of scary. Everybody else we know has recovered. We mostly stay at home. I'm retired so it hasn't affected me very much really. We have started going to the store for food more now. I have about a 1/4 acre garden so during the summer we grow a lot of our own food, but now it is winter and the vegetables don't grow very well. We have a lot in the freezer, but fresh food is nice.

dirkessl, 2020-12-20:
@MartinNohr yes, covid keeps us all busy. I'm sorry for you losing a friend due to it. A close friend of mine also had it and recovered, just his sense of taste hasn't come back, yet. As I live in the middle of Munich, I have no chance of growing my on vegetables, but this would be wonderful if I would be able to do that.

dirkessl, 2020-12-20:
I have put all the info from the Word file into the wiki here, this makes it easier to maintain (at least for me :-) ). Once I will have the hardware running, I will put examples pictures for the inbuilt patterns and other stuff. BOM list needs to be updated.

I have also designed and printed a new case, as on my previous case the TTGO was upside down. Everything fits neatly inside now and there is enough room for the wiring harness. ToDo on the case are snap connectors to make the case close without screws.

I received my aluminum extrusions with a diffuser and they are very low profile, just as I like them. When the hardware runs I will try to put more effort into the gapless connector and look for mounting options for handles and the battery. There is the beauty on open source designs, as the way Martins MIW is constructed does not fit my very minimalistic approach, but might suit others. This way we can supply more options for cases, handles, etc. Just Beautiful.

dirkessl, 2020-12-20:
@DaveStu It works a breeze with the Arduino IDE, as Martin has ironed out all the issues with special libraries. I will provide an detailed instruction on how to install the Arduino IDE and the needed libraries shortly. We will also share an already compiled binary which can be flashed by an inexperienced user with a small tool. We can use the releases feature of github and share a nice changelog.

dirkessl, 2020-12-20:
Documentation is done for compiling it in the Arduino IDE https://github.com/MartinNohr/MagicImageWand/wiki/9.-Compiling-from-source

MartinNohr, 2020-12-21:
I got the BMP preview working nicely now. It's on the first menu. You can use the dial to scroll left and right by screens. It does have a bug for images less than 240 pixels wide. I'll fix that tomorrow, it's past midnight here and my brain goes foggy. :-)
Thanks for all the help guys. This is a great project.

MartinNohr, 2020-12-21:
Preview working fine on everything I tested. I'm sure you'll let me know of any problems. :-)
It's a neat feature.
Dirk: thanks again for the nice image samples. I especially like the space and the flame ones. I had created a simple wall of flame, that was the limit of my talent.

dirkessl, 2020-12-21:
Preview works fine, really amazing. Just I think scrolling is reversed on the wheel. 

My physical build is done, everything works amazingly well. I have some minor issues with the hardware, but will sort them out for sure. I have not run the LEDs on full brightness, yet, but have never needed full brightness on my previous build. Timing on the ESP is much faster which is very nice and will help for sure in the future.

Martin, your talent to code this is out of this world, I am absolutely blown away by the features you have implemented. Truly stunning. I guess it can easily be extended to be of use for other projects, a VILTROX Weeylite RB08P could easily be made for much less money. The TTGO also has a Lipo charging circuit and this would make for a nice affordable RGB LED light. An SD reader is not needed and probably the inbuilt buttons would maybe be enough (but I admit the wheel is very nice) to drive very minimalistic menu system for hue, sat, and brightness + some inbuilt effects (they all have inbuilt effects these days like TV, Police, fire, etc.) A lot of hobby film makers are on a budget...

I will open issues for the bugs, I have found :-)

DaveStu, 2020-12-21:
@MartinNohr Sorry to hear about your loss, it's been an unprecedented year and much sadness. I've had a few friends and ex collegues go through the covid illness, and with some friends of friendswho have been taken away.
Plus UK has got one of the most inept governments in power right now, they've ruined brexit and handled covid so disastrously.

Your fresh foods gardening sounds wonderful, to have the ability. The most I've got to ability to grow are tomatoes and some herbs. The taste experience is very different indeed, comapred to commercial grown produce.
I used to have a friend in Tampa, who bring us florida oranges and admittedly the best oranges I've ever tasted.

DaveStu, 2020-12-21:
@dirkessl Thanks for the uploads. 

Now that I've temporarily connected the parts (haven't got a breadboard), I'll compile/upload the code and give it a test run.
I've been running it through many test programs from various blogs. Had fun playing with all manner of light sequences, testing WIFI which works finding signals, also testing the display but now to some serious stuff.

Look what I found, though not what we need, just found it fascinating. The world of programmable RGB LED's is truly amazing.
https://www.twinkly.com/whats-new/

DaveStu, 2020-12-21:
@dirk I'm not sure how much further your connecting mechanism has got, perhaps you've finished it. I had this idea for the pin connection layout. Apologies for the crude sketching.
[image]
The wires would then be fed through a hole in the extrusion and hidden inside the connection block, with only the pogo style pins showing until connected together. This should allow you to bring the LEDS closer. The data line will obviously have to come from the other side, which can be hidden underneath the LED strip. Maybe use ribbon or FFC cable?

I've pencil sketched something like this, but with a thicker body to hold the two extrusions which I need to sketch in modelling software

dirkessl, 2020-12-21:
@DaveStu Yes, this is basically the idea for the connector, but all very slimline. the things you added will be inside the black parts already.

MartinNohr, 2020-12-22:
Currently the data lines at the junction are not connected to each other, each one goes to a control line on the ttgo. If you want them connected here then we need to run the 3 wires down to the far end of the strip. The software is simplified if we're do that. I originally thought it was clever to avoid the long wire by feeding both strips in the middle. Doing that also means that you don't really need those pogo pins you just need connections to the 3 wires. I'm ok with either way of doing it.

MartinNohr, 2020-12-22:
If you do it my then the pogo pins won't be on the strip buy rather on the carrier housing. Make sense? I'm still trying to avoid that long ugly wire.

dirkessl, 2020-12-22:
I will think about it what's the best way

dirkessl, 2020-12-22:
so I took the MIW for a quick spin to try out the inbuilt patterns. Most of them look really awesome. They look indeed that good that I need a nicer location to shoot them for the documentation. I also found some images do not look really good, as they have no pure black background. I will either fix those or remove them from the standard set

MartinNohr, 2020-12-22:
Excellent! I haven't been out with the new one yet. I've been too busy to even make one yet. I could of course use the images on my older wand to see what they look like. Any kind of non-black that shouldn't be there really does show up. I had that image with some of mine, I just went into Photoshop and used the levels control to clean them up.

MartinNohr, 2020-12-22:
That twinkly site is inspiring. I'm getting some pattern ideas for the built-ins. Thanks, I hadn't ever seen that device. Very clever.

DaveStu, 2020-12-23:
@dirkessl Oh I see, excellent. They always say, if you've thought of it, someone has probably done it ha.

Sorry for my amateur hour, but having problems with the rotary encoder, the switch works but unsure as to where 
+ connection goes too, 5v or 3v rail or other.
I have this type of encoder.
[image]

DaveStu, 2020-12-23:
@MartinNohr I thought you may like the twinkly products, well find inspiration from. Just surprises me what can be achieved.
The possibilities are endless and has given me more ideas, some crazy of course ha

MartinNohr, 2020-12-23:
I don't know about that kind of switch. I designed it around the switch with nothing else. You can see on the schematic how I wired it. Do you have a schematic of that switch board?

MartinNohr, 2020-12-23:
I just expect the switch contacts to connect to ground, no power connections at all.

MartinNohr, 2020-12-23:
In other words I expect the switch to pull to ground. If it does have its own power it would have to be 3.3V. 5V will fry the chip. My worry is that that board inverts the signals. The software can handle that if I put in an option for inverted signals.

MartinNohr, 2020-12-23:
I found the switch schematic. They have pullup resistors on the board. I'm turning on the internal esp32 pullups. You could just ignore the + connection or connect it to 3.3V. Do NOT connect to 5V! It will work fine with no connection. It might have a little better noise immunity with the 3.3V connected. 
Clk=A dt=B and sw is the button on our schematic.

DaveStu, 2020-12-23:
@MartinNohr Thanks, appreciate the help. I updated the code but still needed the 3.3v line, so had to solder that connection.
The switch/encoder works! The way it boots has changed though, without the switch it boots immediately, with the 3.3v line connected, only when I rotate the dial a few times it boots. By design or possibly my switch/encoder type?

Going through your menu myself, I can truly appreciate your intelligent thinking, options galore!
Though I may have a grounding issue or related to switch/enoder.
The LEDS turn on randomly during pattern selection. For example, choosing the solid colour sequence, some LEDS turn on some dont. Cycling through the hues, they are not constant. If for example, I choose 100, then go to another hue then return to 100, the colour will be differernt or some LEDS do not light up.

I'm powering the strip from a usb powerbank and the TTGO from laptop.

MartinNohr, 2020-12-23:
Don't connect the 3.3V line to the switch. I just remembered pulling one of the those lines high will prevent booting. The switch only needs the ground and the contacts connected.

MartinNohr, 2020-12-23:
Interesting about the LEDs. I haven't seen that. I do worry that it might be because our data lines are 3V3, while the WS2812 spec says this is not enough. Try running the LED strip on 4 volts or less, even 3V3 (although it may act funny at that voltage. You may have a strip that needs the 0.7xV on the data line in order to work consistently. That is why my original design included a level converter. There is another way to cover it by sacrificing the first LED on the strip.
Here's the issue and solution. https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/

DaveStu, 2020-12-23:
> 
> 
> Don't connect the 3.3V line to the switch. I just remembered pulling one of the those lines high will prevent booting. The switch only needs the ground and the contacts connected.

Thnaks. I tried that but then only the switch works, not the rotarty controls. The text just flickers when I turn the encoder.

DaveStu, 2020-12-23:
> Interesting about the LEDs. I haven't seen that. I do worry that it might be because our data lines are 3V3, while the WS2812 spec says this is not enough. Try running the LED strip on 4 volts or [...]

I assume the mega2560 output had a higher signal voltage, as the strip worked fine on it?

DaveStu, 2020-12-23:
@MartinNohr Is this something that my encoder would need?
https://github.com/igorantolic/ai-esp32-rotary-encoder

I'm trying not to guess, just trying to learn and get my head around things. Appreciate your help and support.

MartinNohr, 2020-12-23:
I wrote my own dial encoder, you don't need the irgorantolic one. I looked around and didn't like how any of the libraries worked, so I created one with my needs in mind.
The further insight... I just used the raw switch, no board. You can't have those pullup resistors connected to +v, it will mess up the boot logic. Just leave that wire (+V) empty on the board. It should work fine. Isn't it working?

DaveStu, 2020-12-23:
> 
> 
> I wrote my own dial encoder, you don't need the irgorantolic one. I looked around and didn't like how any of the libraries worked, so I created one with my needs in mind.
> The further insight... I just used the raw switch, no board. You can't have those pullup resistors connected to +v, it will mess up the boot logic. Just leave that wire (+V) empty on the board. It should work fine. Isn't it working?
> You can see on the schematic how I wired the switch.

I understand and makes sense. I've been thinking all sorts to figure things out so went all over the internet going crazy.

So, I went back to the mega2560 board to test the strips and they work fine but I encountered a GND issue which lead me to believe a power issue with the TTGO too.
From the USB power bank supplying voltage to the LEDS, I also connected to the 5V and GND board pins.
Effectively removing USB-C to power the board.

Finally the LED's now seem to be behaving, before when turning on the LED intro seqence was crazy, but now it runs smoothly. BUT the encoder problem remains, only the switch works. I can select the first inbuilt pattern (which again confirmed the LEDS working now) but nothing else.
If your's is a RAW encoder, I will remove the resistors. Perhaps they're creating a voltage drop or something else.

DaveStu, 2020-12-23:
Looking at the switch/encoder trace lines, withoutn the + connectors, the resistors wont effect the clk/dt lines.
So pointless removing them. I'm going to connect the GND from encoder directly to power supple. Perhaps my board is faulty
in regards to grounding. 
I think my soldering was bad too. Need a better iron.

dirkessl, 2020-12-23:
I have just unsoldered the encoder from the board as I had exactly the same problems. Works as expected now and also saves space. A Facebook friend of mine (also from the UK), who is also building the MIW did the same and it just works.

DaveStu, 2020-12-23:
> 
> 
> I have just unsoldered the encoder from the board as I had exactly the same problems. Works as expected now and also saves space. A Facebook friend of mine (also from the UK), who is also building the MIW did the same and it just works.

Really!? Very strange considering its just a simple board. I'll do the same then. Thanks, its been very frustrating, plus I must have annoyed Martin quite a bit.

DaveStu, 2020-12-23:
Finally, it works! Thank you both. 
I will add, for me, connecting terminal C of encoder direct to board didn't work, only when connected to negative power it worked, flawlessly. Looking at the traces on a spare encoder, I'm sure If I cut the GND trace and connect a wire from C direct to Negative would have the same effect.

Ok, so now back to thinking of enclosure and attaching designs.

MartinNohr, 2020-12-23:
Good detective work! I had connected my power directly to the 5V pin on the TTGO board. You can't really run the LED's when running from the USB connector, not enough power, strange things happen.
I also suddenly realized what the problem with that board is. If the power is not connected, you have A and B connected through resistors. Bad idea. And if you connect the power, then it messes with the startup conditions. Bad idea.
You could remove the little resistors from that board and it would work. I could also redesign the software to use a different pin so it won't affect the booting, but then that board would have to have power supplied to it so there isn't a bridge between A and B. I could use ESP pins 32-36, but they don't have internal pullups so we would have to always add some pullups for A, B, and the switch. I don't like that either! If you don't mind using the switch without a board that would be my preference.

MartinNohr, 2020-12-23:
Actually if you look at the schematic I uploaded you will notice that I implied a battery connection to the +5V connector and no USB connection. I only use the USB for programming. I'll see if I can make that more clear.

MartinNohr, 2020-12-23:
Hardware guys like to blame software and software guys like to blame hardware. I've always had lots of fun in my career because I've done both at various times.

DaveStu, 2020-12-23:
Your schematic is good, it was just me. I should have realised earlier, after the experience with the Mega2560 board and also as I've fitted so many car audio and amplifiers to classic cars, that grounding/ground loops and power rails was always a problem.
The thing that got me was the encoder GND though, I'm still baffled. From my understanding the resistors would not be involved, unless the + line was connected. It seemed terminal C ground, did not like the GND that was connected to Switch.
Oh well, finally solved thanks to you all.
Had a play around with the patterns. It's amazing how some light sequences can excite haha.
Connected a second strip, all the patterns worked flawlessly, flowing from strip to the other.
Your coding skils are magical @martin. Glad I reached out.

MartinNohr, 2020-12-24:
Dave: Don't worry, I'm not annoyed in the least. I'm always glad to help. Many people helped me learn and understand how things work. I like to give back when I can.
Each of the ESP inputs have an input pullup or pulldown resister (except for 32 to 36) that can be enabled. This allows contacts like switches and dials to be connected with their common parts connected to either + or ground. When you use a pullup, then the switches are wired so they connect to ground. This is very common mostly because it means there isn't any live voltage on the switch and is thus safer because a short to ground won't hurt anything like a short to +v might.
Those resistors on the dial board you used effectively connect A and B together. So if the +V isn't connected then closing A or B would tend to also pull the voltage on the other B or A towards ground. Depending on the value of the internal resistor and the one on the board you might have situation where A and B will follow each other to ground, thus make it not work correctly.
I hope this rambling makes sense. If not I'll try again tomorrow morning when I'm sharper!

MartinNohr, 2020-12-24:
You like working on classic cars? That's cool.
I'm not just a hardware designer and software developer and photographer... I have recently restored a 1953 Ford Tractor, NAA Jubilee edition. It sat in a field for years with a seized engine. It now runs like new. I had parts scattered all over my driveway and garage for a couple of years while I worked on it. My wife kept encouraging me to finish it so she could have the driveway back! I had to take almost everything apart on it. Heavy parts. It was very satisfying the first time I finally cranked it over and it started with that lovely tractor engine sound.
What classic cars have you worked on?

DaveStu, 2020-12-24:
> Dave: Don't worry, I'm not annoyed in the least. I'm always glad to help. Many people helped me learn and understand how things work. I like to give back when I can.
> Each of the ESP inputs have an [...]

Even still, I do really appreciate the guidence and patience. Yes, I do kind of understand it. Some bits are coming back to me.
As I child I was very curious around all things electronics, and would get electronic learning kits as birthday gifts, particulary from Tandy/Radioshack (a long time ago). Like many curious children breaking home appliances (and put them back togther) was my thing, with angry parents of course ha.

DaveStu, 2020-12-24:
> You like working on classic cars? That's cool.
> I'm not just a hardware designer and software developer and photographer... I have recently restored a 1953 Ford Tractor, NAA Jubilee edition. It sat [...]

Sorry, that read wrong. I worked on the audio section of some classic cars. I had some friends, who'd restore froom body work and engines and I'd help them, so I could learn but it was all their domain. I only came in fully, to lay in wiring, fit speakers, amplifiers, sound deadening and the main stereo. Then I'd tune them for optimal sound.
The cars they restored were classic BMW's, Mercede's, some Audi and some VW's, mostly MK1+2 Golfs.

DaveStu, 2020-12-24:
Anyway, Merry Christmas/Happy Holidays to you both, be well and be safe. Will engage here after the festivities.

MartinNohr, 2020-12-24:
Merry Christmas to you as well.
My tractor could use a good audio system. :-)
I've rebuilt a couple of classic VW's as well. 1962 and 1964. Way before the Golf (called Rabbit here).

dirkessl, 2020-12-25:
Merry Christmas to you, I hope you are having a wonderful time with the family!

MartinNohr, 2020-12-26:
Thanks. We're having an interesting time... We drop food and presents at the door for my daughter and grandson. They both have covid so are confined to their house. Slight fevers and tired, but otherwise they're fine. They were in our car two days before they had any symptoms, so we were a little bit worried. We got tested and we don't have it, at least not yet! So we had a very quiet Christmas this year. We did use Skype for opening the presents. We do that Christmas eve, remember I was born in Denmark, so that's my heritage.

MartinNohr, 2020-12-26:
I put up new code this afternoon. I found a font that I think is much better. I had to shorten some of the menu lines, but it is much easier to read. I also added a system setting to change the menu colors. I'm not completely satisfied with how it works, so I will probably change it later when I think of a better way.

MartinNohr, 2020-12-26:
I added the IP address to the main menu so you can find it.
Dirk: Do you mind if I make some changes to the wifi code? I like what you have done so far, but it does need a bit more work. I see the uploading isn't working yet. It would also be nice to be able to get a list of the files down to the web browser. I think this is going to be much better than using Bluetooth. Thanks for your work on this, it's a great start.
I think I'd like to see the settings and be able to change things from the computer/phone web browser. This is going to stretch my knowledge of HTML and buttons and things that I have only a passing familiarity with.

dirkessl, 2020-12-26:
It was a very strange Christmas, indeed. We have a lockdown here in Germany so we are not allowed to be outside between 21:00 and 5:00. The kids were with us and we had a nice evening, but it was strange sending them home at 20:30. If you are outside at night without any urgent reason the police fines you at least EURO 500. My parents in law picked up food and presents from our doorstep as you can only celebrate with one other household so it was really strange not celebrating with them.

dirkessl, 2020-12-26:
@MartinNohr yes please change the code, as I am not very good in programming these advanced features. If you just put basic html code there, I can make it look prettier and responsive if you wish.

I am also thinking about the remote functionality, on how to implement this in a clever way. I guess we have two options here. We can either send commands to a remote unit with a webrequest when a pattern/image runs to a fixed address like http://192.168.4.100/remotestart and http://192.168.4.100/remotestop or we offer an URL which gets pulled by the receiver (http://192.168.4.1/isrunning), but I guess this will harder as the remote receiver needs to constantly pull the website of the MIW which might generate too much load or cause stutter on the MIW. I think I would prefer the first way. The receiving unit could then have a status LED which shows if the unit has received the command or it could use the display if we use another TTGO-Display or the other Board I found with the antenna connector.

MartinNohr, 2020-12-27:
We'll have to do a little experimenting to determine how much load we can put on it. It is a dual processor and pretty fast. Running the leds doesn't use too much computer time. What I don't know is if both of those processors are actually used when we compile single thread stuff. We might have to start using FREERTOS! I'm sure it supports both processors for tasks. It would be fun to have a real OS to work on again. Anyway, just thinking out loud.
It must be strangely quiet in the evening and night there. Almost no traffic, right?

MartinNohr, 2020-12-27:
I noted in the WiFi comments, but I'll mention it here also. I fixed the file upload to SD using WiFi.

MartinNohr, 2020-12-28:
I added a little feature. Pressing button 0 (on the board) will always display the image no matter what else is on the screen. Pressing it again restores whatever was going on before. Note that it doesn't work if an image is actually being shown on the LED strip of course.
I suppose we could make button 0 do other things, but I got tired of selecting a menu item to preview the file.

MartinNohr, 2020-12-28:
Dirk: Would you consider adding the sources for the STL files? Some people might need to modify them slightly for different extrusions.

dirkessl, 2020-12-28:
@MartinNohr sure, but as I use TinkerCAD for all of this, I can only supply online links. I guess I need to learn some Fusion360, I already have a license for it and think I will be much faster once I mastered it.

dirkessl, 2020-12-30:
Quick Update, I took the time today and learned Fusion 360 and created the basic case. I need to print one first, before I can make more adjustments, as I guess I did not include enough tolerances for 3D printing. But it is a good start and Fusion is really powerful.

[23 more replies]

closed 2021-06-02 by MartinNohr