MartinNohr/MagicImageWand #2: gapless connector

dirkessl, 2020-12-15:
Design a connector that doesnt leave a gap and therefore a black line if joining 2 144 LED strips

DaveStu, 2021-11-16:
Hi Dirk, Hope you're well, its been a long while. Any change I can see your gplaess connector please. I've just started back on this project after being shelved for over a year, covid has changed history/life so much. Anyway, I've been trying to figure the option 2 method in martins options to arrange led strips differently to eliminate the gap, unfortunately I cannot get it to work. One strip works, the other doesnt in option 2. I'm also somewhat confused in the wiring the other strip, as it can only be powered in one direction. So, now back to original method, but trying to elimate black space again.

DaveStu, 2021-11-16:
[image]
[image]
[image]

I havent gone down the 3d printed route, after some investigating for a metal off the shelf setup I found a double flash bracket that worked and perfectly holds the alu profile. I mounted a third flash holder in the middle, then screwed in a multi-angle umbrella light stand bracket. 
As the aluminium profile is light weight, this bracket holds them easily and tightly. I Now need to figure out the box for controller and battery, plus wiring. I'm finally getting somewhere.