MartinNohr/MagicImageWand #4: inbuilt remote for camera

dirkessl, 2020-12-15:
Idea: 

Press start on the controller
controller sends signal to receiving unit, triggering bulb mode (or set timer on camera)
controller plays pixels
controller sends signal to receiving unit to close shutter

Either 433 Mhz with a module like this: https://www.aliexpress.com/item/32416047991.html?spm=a2g0o.productlist.0.0.44c933caOL2SaQ&algo_pvid=9416576b-7458-4029-b1be-9e83031af96c&algo_expid=9416576b-7458-4029-b1be-9e83031af96c-4&btsid=0b0a050b16080249160327090ee01c&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

or maybe BLE if range is enough.

highend: LoRa

MartinNohr, 2020-12-15:
This is a wonderful Idea! I like it.
I find that BLE will often go about 10m, but I think that is beyond what the spec says.

MartinNohr, 2020-12-15:
I have also found in testing that the WiFi radio will go closer to 30 meters outside. Maybe even further if the receiver is good.
I designed a radio controlled elevator button and I found it would easily go that far. I put an antenna on the receiver and it worked great, I don't know the exact distance because I didn't want to walk any further away from the house.
So we can probably use the WiFi on the TTGO and design a receiver for the camera. We might need to research how to actually trigger the camera but I hope that information is available somewhere.

dirkessl, 2020-12-15:
triggering the camera is easy with an arduino a resistor, and an 2N2222 transistor. I have built a camera timer based on the M5Stack, the schematics can be found at https://www.essl.de/2020/10/23/timy-the-open-source-camera-timer/ it would be a great addition to it, as the M5 already has WiFi. cheaper would of course be an additional TTGO.

dirkessl, 2020-12-16:
a candidate board for the reciever (or even for MIW): https://de.aliexpress.com/item/10000327309127.html?spm=a2g0o.productlist.0.0.197b4331SJLFWF&algo_pvid=cd2f3d4b-d215-4de5-b461-0e7eeda98eac&algo_expid=cd2f3d4b-d215-4de5-b461-0e7eeda98eac-32&btsid=2100bde116081576784187063e622c&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

as it has an ipex connector for an antenna

zhaoyu3517, 2021-05-09:
I use two esp8266 made a Wireless remote for camera.
use the esp-now:https://www.espressif.com/en/products/software/esp-now/overview
if i play the bmp,then the shutter will be pressed at the same time.

You can understand the principle by watching these videos
https://www.youtube.com/watch?v=w4R9VoY96h8
https://www.youtube.com/watch?v=qxwXwNS3Avw
https://www.youtube.com/watch?v=_cNAsTB5JpM

metalmaslow, 2021-07-28:
i love this  idea!