MartinNohr/MagicImageWand #5: preview of bmp

dirkessl, 2020-12-15:
as the TTGO-T-Display has a nice color TFT display, a preview on the screen would be nice. longer images can be clipped.

MartinNohr, 2020-12-16:
I already did that for the 3" touch screen version I made a while ago. I'll put that code in this version. I had it scrolling sideways for long images. The display only has 135 pixels but we can just trim 9 of them to make the 144 fit.
Good idea!

MartinNohr, 2020-12-21:
Workng nicely now. Not super fast, I will try and optimize it. The dial can move the display left and right by screen size. The sideways scrolling does fail for images < screen size (240 pixels). I'll fix that tomorrow.

MartinNohr, 2020-12-21:
Working well now.

closed 2020-12-21 by MartinNohr