MartinNohr/MagicImageWand #7: Provide a set of default images

dirkessl, 2020-12-16:
Provide a set of images in sizes 144 and 288 ro make for an easy start with the pixelstick

dirkessl, 2020-12-16:
I have already a lot of images for my pixelstick which are royalty free and will categorize them

MartinNohr, 2020-12-18:
I have some as well. Let's create a folder on here to keep them.

MartinNohr, 2020-12-19:
Wow, that's a great collection. Thanks. I have some I did with Mario (Nintendo), slimer (Ghost Busters), etc. I suppose they might be subject to copyright so we shouldn't put them up here. My grandkid liked them. I also have the Millennial Falcon. I've seen that one used before so it might be ok.

DaveStu, 2020-12-19:
You can also grab a bunch from pixelstick website and adjust as needed.

MartinNohr, 2020-12-20:
Just a thought on the fire alphabet...
You can use the macro recording ability to save a macro with a letter sequence. Then you can play back what you spelled. I only support 10 macros now, but that could be increased if we need more. They are numbered 0 to 9. I could easily make it 00 to 99.
Dirk: thanks for the fire alphabet, it inspired me for a great use of the macro recording feature I put in the code.

MartinNohr, 2020-12-20:
I just noticed that the MIW won't read your BMP files. I'll trace in the code and see what is going on. It reads my bmp files ok.

dirkessl, 2020-12-20:
It could be they are in the wrong format, I will check on that. I have created them with a script with imagemagick it might just be a wrong setting in my script.

dirkessl, 2020-12-20:
When I saw the macro functionality, I immediately thought about using it for letters, that's why I made the fire alphabet.

dirkessl, 2020-12-20:
We might want to move the images into a separate repo, as they make this repo very large

dirkessl, 2020-12-20:
> It could be they are in the wrong format, I will check on that. I have created them with a script with imagemagick it might just be a wrong setting in my script.

They are indeed the wrong format (8Bit) I will redo the script and generate them again

dirkessl, 2020-12-20:
The format should be fixed now. I am curently soldering together the new hardware, so I can test as well

MartinNohr, 2020-12-20:
They need to be 24 bit, but they are 24 bit color in the header. What is different is the bitmap info header. It is usually 40 bytes long, yours show as 124 bytes. I can deal with it in the code, but it was something I hadn't seen before. I think there is a color map or something in there. I need to refresh my memory on the BMP format.

dirkessl, 2020-12-20:
Strange, as I am using BMP Version 3 which should have a 40 byte header. The command I use is the following:

filename=$(basename -- $1)
filename="${filename%.*}"
setname=${PWD##*/}
convert $1 -rotate 90 -resize 288 BMP3:../../288/$setname/$filename.bmp

MartinNohr, 2020-12-20:
OK, I see what is going on. Your BMP files are 24 bit, but there is a color table following the DIB header. This is incorrect according to the BMP spec. "present only if Info.BitsPerPixel less than 8  colors". The bits per pixel is in fact set to 0x18. I can easily work around it in the code. There is a pointer to the start of the pixel data, I'll just use that instead of checking the DIB header size. SMOP! (simple matter of programming).

MartinNohr, 2020-12-20:
I used a hex editor to examine the files. (it was the first windows program I ever wrote, it still works!).
I can see the header size set to 0x7c. So something is broken. You can fix it if you want, but I'll still fix the code to use the offset in the BMP header, that is safer coding anyway.

MartinNohr, 2020-12-20:
Interesting. earth1 is bad but earth2 has the correct values.

MartinNohr, 2020-12-20:
OK, I fixed the code to handle those strange bmp files.

MartinNohr, 2020-12-20:
And you fixed the bmp files I see. I looked at a couple, they are standard now.