MartinNohr/MagicImageWand #9: Breadboard testing

MartinNohr, 2020-12-20:
If you are testing this on a breadboard make sure to lower the LED brightness. The power draw with lots of white lights can easily cause a voltage drop and the TTGO will freeze. Or you could also connect a separate power supply to the battery lines on the LED strip. I had a lot of crashes when just using the USB to power everything.
You could just set the brightness lower and then save the system settings with automatic load on.

dirkessl, 2020-12-28:
Interestingly, Is till run everything from the USB-C port and didn't have an issue, yet. I guess it depends on the LED strip, as they are all of mediocre quality. I have the power lines already soldered in, but as I am testing it is just easier to run it from the USB-C connector.

MartinNohr, 2021-01-31:
Well, mediocre quality perhaps, but to be fair, using the strips directly from a 3.3 V control line is out of spec. That isn't the problem I was seeing, but I have seen flickering lights on one strip until I used a logic level translator.