MartinNohr/MagicImageWand #12: when using inbuilt effects bmp preview is still shown

dirkessl, 2020-12-21:
not a big deal, but people might expect a preview of the effect which is of course hard to do.

MartinNohr, 2020-12-22:
I'll block it for the builtins. I couldn't think of a good way to show the builtins on the display. It might be possible though, I'll ponder it after I fix a few of the other issues. The main issue is how to handle the timing. The BMP files are static images while the builtins are mostly dynamic, they change in time. Perhaps I could just show the first 240 frames? It would be useful, you can't tell what they look like by displaying them until you take a picture of them.

MartinNohr, 2020-12-22:
Also let me know if you have any good ideas for more builtins.

MartinNohr, 2020-12-22:
eingebaut, (danish inbyggede), but the real english is reversed: built-in, not inbuilt :-)

MartinNohr, 2020-12-22:
I fixed this. This shows how elegant my menu system is (well, at least in my opinion). I just had to move one line in front of the eEndif instead of after it.
    {eIfEqual,false,"",NULL,&bShowBuiltInTests,true},
        {eBool,false,"Switch to SD Card",ToggleFilesBuiltin,&bShowBuiltInTests,0,0,0,"On","Off"},
    {eElse},
        {eBool,false,"Switch to Built-ins",ToggleFilesBuiltin,&bShowBuiltInTests,0,0,0,"On","Off"},
        {eText,false,"Preview BMP",ShowBmp},
    {eEndif},

closed 2020-12-22 by MartinNohr