MartinNohr/MagicImageWand #14: Reverse Display not working correctly

MartinNohr, 2020-12-21:
It gets confused about what is normal or reverse.

dirkessl, 2020-12-21:
also, yellow on white is hard to read.

I think normal should be black background, as the device gets mostly used at night and a black background is easier on the eyes

MartinNohr, 2020-12-22:
Maybe we should just remove this option. I don't find it useful either. I was just calling the TFT library call "reverse". It doesn't seem to be very predictable. I can see changing the character colors and brightness, but the background should probably always be black.

MartinNohr, 2020-12-22:
Aha, it turns yellow because I had the foreground color set to blue. Blue and Yellow are opposites.

MartinNohr, 2020-12-22:
Reverse display was better when there weren't any colors.
I think maybe always black background and allow two foreground colors, active and inactive,  like in the menus. Agree?
Or we could allow some dark backgrounds, like dark grey, but that is probably me overthinking this.

dirkessl, 2020-12-22:
I think it is safe to remove this option

MartinNohr, 2020-12-23:
Gone. I will add settings for the menu colors soon.

closed 2020-12-23 by MartinNohr