MartinNohr/MagicImageWand #16: Option to double font size

dirkessl, 2020-12-21:
Or as you already said, make the font a bit larger

MartinNohr, 2020-12-22:
Yes, we definitely need something. When my eyes get tired it is getting hard to see. It will mean fewer lines on the display, but I don't think that is a problem. I'm more worried about the width/length of the lines. I'll come up with something.

dirkessl, 2020-12-22:
maybe even 1.5 times larger is enough, long line could scroll if staying active (top line) for 250 msec.

MartinNohr, 2020-12-28:
I came up with a much nicer font. It is larger, but not too large.
The font size control in the TFT library only allows integers, so 2x would be the next size. This severely limits the number of characters that can be displayed in the menu. We could of course scroll sideways but that will really slow things down.
Try the latest code and let me know if that is acceptable.

dirkessl, 2020-12-28:
It looks really good and very readable now. I will close this as I can even read it without glasses. Well done, Martin, I see you have great taste for fonts.

closed 2020-12-28 by dirkessl