MartinNohr/MagicImageWand #20: Fade In / Fade Out

dirkessl, 2020-12-22:
Some Patterns or images might benefit from a fade in / fade out option

MartinNohr, 2020-12-23:
Indeed they would. I thought of putting in an "envelope" to allow brightness shaping based on a grayscale image, but maybe that is too complex. It would be pretty easy to add a fade in/out based on a time or frame count setting. Would that be good enough?

dirkessl, 2020-12-23:
Frame based would be harder for the internal patterns, I guess, so I would vote for time based

MartinNohr, 2020-12-23:
It is much harder to do for the internal ones. They are all kind of stand-alone functions. The BMP display function can easily handle it. Although... thinking... maybe I could use an interrupt timer to do it. Might be kind of messy. I'll give it some thought this evening.
I'll come up with something, I think the fade in/out is a good idea. Shall we have adjustable start/end or is black always that value. I vote for black.

dirkessl, 2020-12-28:
another vote for black.

MartinNohr, 2021-01-02:
I added a fade in/out count to the  "file Image settings" menu. It is the number of frames to fade and out. 0 to 255 is allowed. I could make it longer but it would require floating arithmetic and I don't think it is really necessary. GIve it a try.
I haven't figured out any way to add it to the built-ins. It is of difficult to fade out the ones that run forever, but I suppose it could be done when the long press cancels the playing. I can probably figure out a way to do it. I'll keep giving it some thought, but for now it only works with BMP files.

MartinNohr, 2021-01-02:
After I wrote the above I thought of a reasonable way to do the fade in/out for everything. It's a bit of work, but not too bad. I'll probably work on it next week.

MartinNohr, 2021-01-02:
I have updated the code with improved handling of the fade in/out. It should now be possible to add fading to the built-ins. I haven't done this yet, but I will work on it soon.

MartinNohr, 2021-01-02:
Re-readding this I noticed you liked the time based setting. Frame/column based was a lot easier to code. I could still make it time based, but I would have to map the time onto columns. I'm not sure that would be any better. It could also be a percentage of the total size, that almost makes more sense. For example: 10% of the total width. Anyway, let's play with the frame count based one and see if that works ok.