MartinNohr/MagicImageWand #21: WiFi Image upload

dirkessl, 2020-12-22:
when in the field without a computer/SD reader it could be helpful to upload images over WiFi. Images can be converted with the online converter beforehand.
Care must be taken to not collide with the planned remote shutter functionality.
Idea: MIW will create a Hotspot and will offer a small website for uploading images and expose an API where the remote shutter can connect to. This will be a bit more work on the remote shutter, as it needs an UI to connect to the MIW

dirkessl, 2020-12-22:
I have started on the WiFi code, testing if it compiles. It seems with BLE enabled it did not work as the WiFi module was suing the same dram space. Now with BLE removed it creates an AP which you can connect to. Removing BLE also brought back a lot of program storage space.

MartinNohr, 2020-12-23:
BLE is a memory hog, mostly because I think it includes all of bluetooth classic. I'm surprised you can't use both, but I never tried that.

MartinNohr, 2020-12-23:
I'm not sure BLE is very useful, but wifi is probably extremely useful. 
My wifi skills are weak, I appreciate greatly the work you do on this. Thanks.

dirkessl, 2020-12-23:
I am having a hard time implementing it :-) I will try further after the Christmas

dirkessl, 2020-12-25:
@MartinNohr I have implemented preliminary WiFi functions, I just can't get the upload / download stuff not to work If you could have a look into those functions, that would be awesome. I also don't know where the uploaded files should go to, maybe into an additional wifi folder? I am sure you have some ideas on this.

MartinNohr, 2020-12-26:
I'm done with other stuff for now so I will take a look at this soon.
BTW, I notice you're using F("..."). It doesn't hurt anything but it isn't necessary on the ESP32. The old arduinos had 2k of ram, esp32 has 520k built-in and usually 4M external. There's plenty of memory now.
You might have done some work on the download already, I just tried it and it worked, the only problem (at least with Android) is that .phps extension gets appended to the filename and has to be removed in order to use the file normally.
Uploading files doesn't work, I'll fix the .open call. I wish the SD library worked like standard C file IO, but it doesn't work the same, similar but not exact. I found a library to implement standard C file IO calls on the esp32 but I haven't found all the pieces to make it work yet. Future project.
There should also be a function to get the SD file list back to the web browser so the user can see what files are on the SD card.

MartinNohr, 2020-12-27:
I fixed the file upload. You can now upload files. It does need some improvements, like handling a filename that already exists.
We also need the ability to show a list of existing files on the SD card. I'll see if I can figure out how to do that.

MartinNohr, 2020-12-27:
Other features:
- show a list of existing file, maybe from a button press.
- show settings and allow starting playback.

MartinNohr, 2020-12-27:
I figured out how to show a list of the files on the download page, but I don't know enough HTML to figure out how to make them clickable for selection.

MartinNohr, 2020-12-27:
I added a couple of settings on a new page. Not very fancy, I need to learn a lot more about HTML. Like, how to change those settings, and a prettier display.

dirkessl, 2020-12-27:
Nice work, I tried the upload function and it works, but the uploaded picture does not show on the MIW. Filename when uploaded from iOS is something like FDB159FC-2E4E-43EE-A48E-59C14864F147.bmb so that's very long and very practical. Maybe we should add another textfield to the upload function where one can type in a desired name. If no name is given, it reverts back to upload + random(100,99999)

I saw the new settings page, really great. how would you want to change the settings there? with up/down arrows? I could make the buttons and you provide the code for changing the values. A simple text field would of course also work.

MartinNohr, 2020-12-28:
I forgot to reload the file list. If you reboot, or change folders and then come back the file will appear.

MartinNohr, 2020-12-28:
There are many more settings to implement. We should consider how to organize them so we don't make a complicated mess. Perhaps something along the same organization as the menus on the Wand would be good. Spin buttons would be nice but manual entry is acceptable to me.
I've been reading about how html works. I'm slowly starting to understand.
I think I might even know how to make the filenames selectable from the list, and to change folders! 
We should be careful at this point that we don't start working in the same areas of the HTML code, we might have a lot of conflict resolution to do when we check in changes!
If you could code a sample spinner button for the column time I can probably figure out the others.
Some of the settings are also check boxes.
Supper time here, must go eat!

dirkessl, 2020-12-28:
> I forgot to reload the file list. If you reboot, or change folders and then come back the file will appear.

Martin, try yourself: rename a bmp to FDB159FC-2E4E-43EE-A48E-59C14864F147.bmb and it wont show up in the list.

MartinNohr, 2021-01-18:
Interesting. I'll look at that as soon as I can. I teach a digital photography class at the university here, and I need to finish my lecture notes for tomorrows first class. Sorry I just noticed now, so late.
BTW, there were some problems with long names that I fixed, so this might have gone away already.

MartinNohr, 2021-01-22:
There was a 39 character filename length limit in the code. I changed it to 99. Your long name works fine now.

MartinNohr, 2022-09-18:
I have fixed many of the above issues with upload/download. The settings page now allows text entry to change various settings. I don't think we need to have all of them, but it could be done. The code is list driven so we can add whatever controls we want to show up on the web settings page without typing a lot of HTML. It doesn't have menu levels like the MIW, just titles between sections. Values are all shown as text or checkboxes. I suppose spin buttons might be useful, so might sliders also. I'll think about that later.
Up and downloading files is much better now, the only missing thing is the ability to select the upload folder, it always puts them in the root. I will change that when I have some more time. I've learned enough HTML to do some of these things
Try the latest version and let me know what you think.

MartinNohr, 2022-10-04:
I've now improved the server pages a lot. What do you think?