MartinNohr/MagicImageWand #22: Directory up should be on the top

dirkessl, 2020-12-22:
If you jump inside a directory with a lot of files by accident, going back can only be achieved by scrolling completely to the bottom. Workaround at the moment: turn on menu wrap

MartinNohr, 2020-12-23:
Ok, I had it that way in the past, but I can't remember why I changed it. I'll make it first again.

MartinNohr, 2020-12-23:
Any idea for a better symbol,  or is ^ ok?

MartinNohr, 2020-12-23:
I changed the symbols to < and >. Seems to make more sense to me.
Back up folder is now sorted to the beginning of the list.

closed 2020-12-23 by MartinNohr

dirkessl, 2020-12-23:
Great stuff! 

—
Dirk
www.essl.de | www.lenstagger.com | www.caffenol.org

MartinNohr, 2020-12-23:
Thanks to your comments this thing is getting better and better. I'm very happy to have others involved in improving it. :-)