MartinNohr/MagicImageWand #24: Standard Direction should be right to left

dirkessl, 2020-12-22:
When recording macros, for instance spelling a name with the fire alphabet, names need to be put in reverse in the macro menu as walking direction is reversed.

I would suggest to change the standard direction to right to left, this would also make all pixelstick images on the web compatible without rotating them 180 degrees.

Or am I missing something?

MartinNohr, 2020-12-23:
Nope, I missed that one. It happened because the original design moved from right to left. I decided that it seemed more natural to move from left to right. It also made reading the images faster since I didn't have to read them in reverse, because of rotating them clockwise. Perhaps we should rotate the images counter clock wise instead of clockwise and make the default walking from right to left. I'll sleep on that idea, we can decide tomorrow. I am thinking it might be better to do it that way. Or else we have to explain to people to record macros in reverse! Probably silly.

MartinNohr, 2020-12-23:
Rotating CCS means flipping top and bottom during display, but I already handle that in code and it doesn't have any time penalty. Reading the file in reverse does have a time penalty. So I am thinking we should go with CCW rotation. I seem to remember that the original Michael Ross version also rotated CCW. You'll have to change your website converter but I suspect that will be pretty simple.

MartinNohr, 2020-12-23:
OK, after some thought... let's reverse all the images to CCW rotation. I'll fix the code this morning and change the docx file to document the correct direction etc.

MartinNohr, 2020-12-23:
Latest code now works as discussed above. Images need to be rotated CCW and walking is from right to left as facing the camera. You can still reverse the walk direction, there is a slight time penalty, but it isn't too bad, maybe 20% (guessing...).

closed 2020-12-23 by MartinNohr