MartinNohr/MagicImageWand #25: Play inbuilt patterns forever until long press

dirkessl, 2020-12-23:
It would be nice for the inbuilts to play forever until long press, as the patterns all have different play times which is hard to keep track off. Or make it possible to save MIW files for inbuilts.

MartinNohr, 2020-12-23:
Yeah, they are kind of a hodge-podge. Perhaps running forever would be best.

MartinNohr, 2020-12-24:
After some thought, I agree. I'll remove the timers from the built-ins. Unlike BMP files, the built-ins don't really have a clear ending point, so they might as run until they are told to stop. Thanks for the idea.

MartinNohr, 2020-12-24:
I have changed the built-ins to run forever for the ones that had timers. Some of them ran just once normally, I have left those running just one time, but they can of course be repeated by setting the repeat option in the repeat settings menu.
Try them and let me know how they work for you.

dirkessl, 2020-12-28:
cyclone, meteor, one dot, rainbow pulse, two dots, wedge run only once. We either need an indicator in the menu for single running or forever running patterns, or they all should run forever. Also the single runs do not show a progress bar or time left.

MartinNohr, 2020-12-29:
Any of those can of course be made to repeat by setting the repeat count. Still, you raise some valid issues. I'll work on it.

MartinNohr, 2020-12-29:
There are actually two kinds of built-in patterns.
1. Some don't have a natural stopping point, like checkerboard or barber pole, etc. These make sense to run forever. There should be a message on the screen to use long-press to stop. I had put a timer on these to make them stop eventually, the long press is fine to stop these.
2. Some patterns have a natural stopping point, like wedge or running a dot from one end to the other. I think it makes sense to only allow these to run once. They can always be set to repeat with the repeat menu entry. If you always have them self repeating then the natural stopping point doesn't mean anything, and if you wanted to stop at that point it will be very difficult to do with the long press.

So after some thought I think we need to have both kinds, but I agree that it should be made more clear. I could add something to the name that indicates once or forever. An ∞ symbol would be nice but the TFT fonts don't have one. I could probably create my own though. I've never done that but I have read that it is possible.

MartinNohr, 2020-12-29:
BTW, it isn't "cyclone", it is Cylon. It's a reference to the old movie/TV series Battestar Galactica. It was also used for the Kitt car in Knight Rider TV series. They were both created by Glen Larson (I think that was his name without looking it up).