MartinNohr/MagicImageWand #28: ORB painting

MartinNohr, 2020-12-28:
ORB painting is done by swinging a light around in circles. I keep thinking there might be some way to do that with this wand and a spinner handle. We'll need to turn certain lights on on each end and maybe let them change over time. Anyway, something to think about when bored!

dirkessl, 2020-12-28:
This would need additional lights on the very end of the strip (top and bottom) as if you spin orbs and pint the stick sideways and backwards, the LEDs of the strip can not be seen anymore. also the display needs to be off, as it would show in the orb.

MartinNohr, 2021-01-02:
True, not trivial, but it was an interesting idea. Perhaps someday when we have nothing else to do!

tobias966, 2022-01-20:
I tried to use the MIW with single leds WS2811. Worked perfectly. When I use it in lightbar mode and put only 10 led in a ball, I can made perfectly orbs. It would be nice, if you could add an additional function in lightbar mode, so that I can adjust for 2nd controller a different color, then I could made orbs with two colors.

MartinNohr, 2022-01-23:
Cool! Orbs are fun. Send me an image.

Anyway, yes, I can probably come up with some way to do that. The code will be easy, the button interface a bit trickier. I’ll have to think of some easy way to select different colors for the two parts. It might be a better idea to split the display in half with two different colors. Then it could be done with a single controller as well.