MartinNohr/MagicImageWand #34: Control of the stripes

firefighter2608, 2021-01-30:
I would like to push the two stirpes completely together in the middle and start the lines for the stripes once from above and the other from below. then I have no gap in the middle. can ma set this in the code so that it rotates the control by 180 degrees or something like that?
I would appreciate an answer.
Thanks in advance
Peter

MartinNohr, 2021-01-31:
Should be possible of course. I'm busy for the next couple of days, but I will take a look at it as soon as I can. Thanks for the suggestion. You're talking about the built-in stripes pattern, correct?

firefighter2608, 2021-01-31:
Super thank you.
yes it's about the black stripe between the upper and lower stripe. Then you could connect them without a gap and then control them from above and below.

MartinNohr, 2021-01-31:
Can you draw me a picture of what it would look like to make sure I understand correctly?
Thx

firefighter2608, 2021-02-02:
I hope the drawing is understandable.
My idea would be to reverse the LED control in the software, change from current LED1 to LED144. this for every stripe, then you could connect the stripes without a gap but electrically separated and the black line would be gone.

[image]

MartinNohr, 2021-02-03:
Aha, I misunderstood at first, now I see what you are trying to do. I originally did that, but I didn't like running the wires down to the ends of the strip. However, if you don't care about that then this is easily done. Here's the code that handles it.
This would need to be changed so that they are reversed. I can change it for you but it will be a couple of days before I get around to it. In any case, the mapping is all handled in this one routine which should make it easy to change. 
int AdjustStripIndex(int ix)
{
	if (ix < NUM_LEDS) {
		ix = (NUM_LEDS - 1 - ix);
	}
	ix = max(0, ix);
	ix = min(STRIPLENGTH - 1, ix);
	return ix;
}
This just needs to be changed to reverse the mapping to each strip.
I think... I'll consider it later this week when I have some more time to think about this carefully. But I think it will be easy to change.

MartinNohr, 2021-02-03:
Actually, this might do it...
int AdjustStripIndex(int ix)
{
	if (ix >= NUM_LEDS) {
		ix = (NUM_LEDS - 1 - ix);
	}
	ix = max(0, ix);
	ix = min(STRIPLENGTH - 1, ix);
	return ix;
}

MartinNohr, 2021-02-03:
I think the above will work, all you do is change the < to >=. I'll test it later this week if you don't have time to test it. I think it will work, but sometimes I make mistakes. :-)

MartinNohr, 2021-02-03:
BTW, I think you have a really good idea. Without those wires you can easily mash the led's close together and get rid of the black stripe down the middle. I designed a clip to hold the wires to the aluminum extrusion, but I didn't like having that wire running down to the end. But maybe having that is ok given that the black stripe disappears. Next time I work on the software I might add this as an option so it will be easy to change for either wiring method.
Thanks for the idea. I appreciate it when people suggest good ideas. I never think of everything.

firefighter2608, 2021-02-03:
great, thank you very much for the work. the stick works great. is already my third project which I am recreating, but this one is really perfect. Thanks a lot for this. i will test it on the weekend. the wires can be hidden in my profile on the sides. I will report it.

MartinNohr, 2021-02-04:
I'm going to put a new option in the code. I think there are three possible connections that make sense. 
1. Both data inputs in the middle.
2. Both data inputs on the outside. (your idea)
3. Data input for stick 1 on the bottom, and the output from 1 connected to the input of 2, set the number of pixels to 288.
I'll add this option to the menu in the next couple of days.

firefighter2608, 2021-02-04:
I'm going to put a new option in the code. I think there are three possible connections that make sense.

Both data inputs in the middle.
Both data inputs on the outside. (your idea)
Data input for stick 1 on the bottom, and the output from 1 connected to the input of 2, set the number of pixels to 288.
I'll add this option to the menu in the next couple of days.

MartinNohr, 2021-02-05:
Looks like you quoted me, was there a comment also? I'm not seeing it.

ThomasKlee64, 2021-05-29:
Hello Martin,

i have a problem with the second LED Stripe, this one unfortunately does not work. I have selected in the menu under LED Strip Settings Total LED 288 and under LED wiring Mode 2. Unfortunately the second stripe is dead. Both LED strips are connected to the data pin input, is that correct?

What could I have done wrong here?

THX

Thomas

MartinNohr, 2021-06-05:
The code now has code for the three different wiring options. It is working, I'll close this now.

closed 2021-06-05 by MartinNohr

MartinNohr, 2021-07-08:
I just noticed this while cleaning my email. If I never answered… sorry.
Is this still a problem, or did you figure it out?
There are two data “output” pins. One for each strip.
Are you using my PCB, or did you make your own circuit?
If you respond to ***@***.*** I will probably respond faster! This email is getting too much spam.
Martin

From: ThomasKlee64
Sent: Saturday, May 29, 2021 11:11 AM
To: MartinNohr/MagicImageWand
Cc: MartinNohr; Comment
Subject: Re: [MartinNohr/MagicImageWand] Control of the stripes (#34)

Hello Martin,
i have a problem with the second LED Stripe, this one unfortunately does not work. I have selected in the menu under LED Strip Settings Total LED 288 and under LED wiring Mode 2. Unfortunately the second stripe is dead. Both LED strips are connected to the data pin input, is that correct?
What could I have done wrong here?
THX
Thomas