MartinNohr/MagicImageWand #35: M5 Core2 Port

MartinNohr, 2021-02-17:
I could probably make it work on that m5 stack. I thought of using one once. I would have to get one and see how hard it would be. And I would have to find the time to test it. If you want to keep it simpler I do have a PCB made with the TTGO and an SD card reader attached. 
Anyway, I'll go look at the m5 stack specs to see how easy it might be to port the code.
I've been to Norway a few times. I was born in Århus and until I was 18 I lived in Ryomgård Denmark.
Jeg kan godt læse norsk, dog ser det ud som om der er mange stavefejl. :-)

_Originally posted by @MartinNohr in https://github.com/MartinNohr/MagicImageWand/issues/1#issuecomment-770267243_

MartinNohr, 2021-02-17:
I have an M5 Core2 now. I'll port the code as soon as I have some free time.

MartinNohr, 2021-03-25:
I almost have the port to the M5 working. I'll create a new github entry for it as soon as I'm satisfied. It was a bit of work, the screen is different and I decided to enable touch detection. The touch detection isn't great, the screen is small, but it does work. It might be good to still use the dial, it is in many ways easier to use, especially with gloves one.
I'm thinking of using the level sensor on the M5 to show if you're holding the wand level! I might even use it so you can shake the wand to start it playing.

MartinNohr, 2021-04-18:
I'm not finding the touch screen on the core2 to be very reliable or useful. So I have ported the code to the original M5Stack with 3 physical buttons. This is proving to be quite good. I'll post the code when enough of it is working and put a link here.

dirkessl, 2021-05-27:
That is awesome news, it would make it possible to make a MIW without any soldering (when skipping the rotary dial) as it has enough connectors free to plug the LED strip and the rotary dial, making it very easy for even more people building one. I have the older version without the touchscreen and use it as a timer and absolutely love it.

MartinNohr, 2021-06-02:
https://github.com/MartinNohr/MagicImageWand-M5Stack/tree/master
Macros aren't working yet, and the calibration for the level sensor is hard-coded.
Please try the code.
There is some soldering, but not much. The display is beautiful.
I'll add some pictures of the wiring harness that I made.

MartinNohr, 2021-06-02:
I haven't put the rotary dial in yet. Also, I think the M5 SD code only handles <=32G cards.

MartinNohr, 2021-06-05:
I made a mistake. The link I sent you above was for the M5 Stack new version with the touch screen. I didn't like it and never finished it.
Here is the one for the original M5 Stack with the 3 buttons. It is the grey version, but might work on the white one also.
https://github.com/MartinNohr/MagicImageWand-M5StackGray