MartinNohr/MagicImageWand #36: A few errors....

krealab, 2021-02-20:
Hi! I just ordered the TTGO board and downloaded the code. Followed the instructions in the wiki, but ending up with a few errors..... : 

Arduino:1.8.13 (Mac OS X), Kort"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from sketch/MagicImageWand.h:52:0,
                 from /Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Desktop/MagicImageWand/MagicImageWand.ino:7:
/Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Documents/Arduino/libraries/FastLED/src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
In file included from /Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Documents/Arduino/libraries/FastLED/src/FastLED.h:65:0,
                 from sketch/MagicImageWand.h:52,
                 from /Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Desktop/MagicImageWand/MagicImageWand.ino:7:
/Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Documents/Arduino/libraries/FastLED/src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool
Mange bibliotek ble funnet for "FastLED.h"
I bruk: /Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Documents/Arduino/libraries/FastLED
Ikke i bruk: /Users/kolbjorn/Dropbox/Min Mac (Kolbjørns MacBook Pro)/Documents/Arduino/libraries/FastLED-master
Mange bibliotek ble funnet for "WiFi.h"
I bruk: /Users/kolbjorn/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
Ikke i bruk: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returnerte 255
Feil ved kompilering for kort ESP32 Dev Module.

dirkessl, 2021-02-20:
Have you done all steps as in https://github.com/MartinNohr/MagicImageWand/wiki/9.-Compiling-from-source
?

krealab, 2021-02-21:
Yes. I really dont know what the problem is. But i will try a different computer on windows. This i tried on a Mac. This error keeps popping up: enumeration value 'eIfEqual' not handled in switch [-Werror=switch]

Kolbjørn

MartinNohr, 2021-02-22:
First, ignore the pragma messages, they do not indicate a problem. They are just information.
I'm not sure what the Mac messages mean, but I can read the norwegian. It appears among other things that it is finding more than one version of the fastled.h library. Interesting. Are you using arduino IDE? Have you set the correct TTGO device?
The 'eIfEqual' is defined as an enum in the .h file. I do not know why that would cause a compile error. It is a constant value used in a switch in the code, but other constants are also used in switch statements. I have a Mac here, but I have never used it to compile Arduino code.

krealab, 2021-02-23:
I did the same process on a windows computer. It all works now. So now i need the hardware. Do you sell it or should i get it from Amazon? Kolbjørn

MartinNohr, 2021-02-24:
Which hardware? There are lots of parts! The led strips are available from Amazon or banggood. I think I listed them in the docx file.
You can build your own ttgo connections but it you like the PCB version, it is available from me. I have designed a new version of the 3D case, but I haven't uploaded it yet. The PCB has a socket for the TTGO chip and all the components including the rotary switch are soldered on. I can sell the PCB for $20 without the TTGO chip. You'll have to pay some shipping of course.

MartinNohr, 2021-02-24:
Here are photos of the PCB.
[image]
[image]

MartinNohr, 2021-02-24:
There is a newer design with some minor changes that you will be able to order directly from PCBWAY.com in China. It is about a month away from being finalized. I'll post the link when I have tested the prototypes.

ahennies-22, 2021-04-14:
Is there still a possibility to get an assembled board? If not, is there already news about the new board?

MartinNohr, 2021-04-14:
Yes, I have 16 PCB's available. $35 + shipping. This includes a programmed TTGO. I just need an address and I can get the shipping cost.
You will be able to get the board from PCBWAY.com without the TTGO. It will be available soon. I'll send the link when it is ready.

ahennies-22, 2021-04-14:
@MartinNohr 
Hallo,
die Anschrift lautet:
Achim Hennies
Klappenbergweg 14
37176 Nörten-Hardenberg
Kann ich vielleicht per PayPal Freunde bezahlen?
Schöne Grüße
Achim ahennies-22

MartinNohr, 2021-04-14:
martinnohr76@gmail.com
Let me know when you send an email. I don't want to leave this email public for very long. I'll delete it after you have it.

ahennies-22, 2021-04-14:
You can delete it

MartinNohr, 2021-04-29:
> Yes. I really dont know what the problem is. But i will try a different computer on windows. This i tried on a Mac. This error keeps popping up: enumeration value 'eIfEqual' not handled in switch [-Werror=switch]
> 
> Kolbjørn

I think I may have solved this compile problem in the latest version of the code. Some compilers issue a warning/error if all of the enums are not handled in a switch statement. This is perhaps a good idea since it will warn you about missing case statements. Please try compiling it on the system that gave the error and let me know. Thanks.

ahennies-22, 2021-04-29:
I am very sorry, but I still get this known error message. For more details
I have put the output of the compilation into a file. I hope it helps.
The compile process stops in line 334.
Achim

Arduino: 1.8.13 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/User_Setup_Select.h:133:0,

                 from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:39,

                 from sketch\MagicImageWand.h:61,

                 from D:\Pixelstick\MagicImageWand\MagicImageWand.ino:7:

C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/TFT_Drivers/ILI9341_Defines.h:4:0: warning: "TFT_HEIGHT" redefined

 #define TFT_HEIGHT 320

 ^

In file included from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/User_Setup_Select.h:22:0,

                 from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:39,

                 from sketch\MagicImageWand.h:61,

                 from D:\Pixelstick\MagicImageWand\MagicImageWand.ino:7:

C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/User_Setup.h:83:0: note: this is the location of the previous definition

 #define TFT_HEIGHT 240 // GC9A01 240 x 240

 ^

In file included from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:78:0,

                 from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:52,

                 from sketch\MagicImageWand.h:61,

                 from D:\Pixelstick\MagicImageWand\MagicImageWand.ino:7:

C:\Users\hennies\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/FS.h:30:0: warning: "FILE_READ" redefined

 #define FILE_READ       "r"

 ^

In file included from C:\Users\hennies\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatFile.h:797:0,

                 from C:\Users\hennies\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatVolume.h:28,

                 from C:\Users\hennies\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatLib.h:27,

                 from C:\Users\hennies\Documents\Arduino\libraries\SdFat\src/sdfat.h:33,

                 from sketch\MagicImageWand.h:34,

                 from D:\Pixelstick\MagicImageWand\MagicImageWand.ino:7:

c:\users\hennies\documents\arduino\libraries\sdfat\src\common\arduinofiles.h:31:0: note: this is the location of the previous definition

 #define FILE_READ O_RDONLY

 ^

In file included from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:78:0,

                 from C:\Users\hennies\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:52,
[... 667 lines cut ...]
D:\Pixelstick\MagicImageWand\MagicImageWand.ino:3430:20: warning: 'ix2' may be used uninitialized in this function [-Wmaybe-uninitialized]

   leds[ix2] = pixel;

                    ^

D:\Pixelstick\MagicImageWand\MagicImageWand.ino: In function 'void MacroLoadRun(MenuItem*, bool)':

D:\Pixelstick\MagicImageWand\MagicImageWand.ino:3303:3: warning: 'oldShowBuiltins' may be used uninitialized in this function [-Wmaybe-uninitialized]

   if (oldShowBuiltins != bShowBuiltInTests) {

   ^

cc1plus.exe: some warnings being treated as errors

Mehrere Bibliotheken wurden für "WiFi.h" gefunden

 Benutzt: C:\Users\hennies\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

enumeration value 'eIfEqual' not handled in switch [-Werror=switch]

MartinNohr, 2021-05-10:
I fixed the compile errors about the "eIfEqual" giving an error. It should also work correctly on the Mac version now.

ahennies-22, 2021-05-17:
I apologize for getting back to you so late. Unfortunately I was very busy
and had no time to test the new version. Now I have tested it, but
unfortunately the error still comes again.
I will reinstall my computer (I wanted to do that anyway) and then test it
again. But that will probably not happen until next week. I still think
that it is not your code, but my computer. I have had this computer for a
long time and I have installed and uninstalled many things. So, Windows
doesn't like that very much.
When I'm done, I'll get back to you.

Translated with www.DeepL.com/Translator (free version)

ahennies-22, 2021-05-31:
Hi, I have now tested the code on a newly installed Windows 10 computer and
it compiles without problems. Then I tried another test on a windows 10
laptop and it works fine too.
I'm glad that it works now and I'll be sure to keep checking github to see
how this really fantastic project develops.
Thanks a lot for the great support.
All the best

MartinNohr, 2021-07-08:
I have a new design of the PCB that is much better. The signals are now stronger and reliable to the LED's so you can have longer lines without any problems. To anybody that has my original board I can ship you a new one for the cost of shipping. It is the same size.

MartinNohr, 2021-11-24:
I fixed the original errors in the code, it was caused by the compiler being set to check that all enums were used in a switch. Some of them weren't in the switch, not a problem, but the compiler wasn't happy.

closed 2021-11-24 by MartinNohr