MartinNohr/MagicImageWand #37: I made two more

zhaoyu3517, 2021-04-21:
[image]
[image]
[image]
[image]
[image]

This use two 18350 battery. One 144 LED strips.
[image]
[image]
[image]

[image]
[image]
[image]
[image]

[image]

This use 4 18350 battery. Two 144 LED strips.

[image]
[image]
[image]
[image]
[image]

MartinNohr, 2021-06-05:
Very nice mechanical design. I am impressed.
I have added many improvements to the code, take a look when you have time.
One of the things I added is an option to display the battery voltage. I use two 18650 on mine and 20k/10k voltage divider to drop the battery voltage down to <3.3. With proper calibration you can then an approximate % battery.
I also added an option to control the maximum current that is allowed.
Many other changes in the last couple of weeks also.

MartinNohr, 2021-06-05:
The menus and filenames now automatically scroll sideways if they are too long for the screen.

MartinNohr, 2021-07-08:
You have inspired me to improve my own mechanical and electrical design. I modified the PCB so it now has a power supply on the PCB. This makes hooking things up much easier. It is only 2Amp but I have seldom needed even that much during night photography.
I have also added a light sensor, which could be used for remote IR triggering or to automatically control the brightness. I don't know if that is useful or not, I'm just experimenting. :-)
You have the best mechanical design I have seen, thanks for sharing the photos.

zhaoyu3517, 2021-07-15:
I saw your design, you do better. :D

metalmaslow, 2021-07-28:
wow looks awesome!

codenamekraken, 2024-12-19:
> I saw your design, you do better. :D

can you plese help me i also want to make this

codenamekraken, 2024-12-25:
> Very nice mechanical design. I am impressed. I have added many improvements to the code, take a look when you have time. One of the things I added is an option to display the battery voltage. I use [...]

> [image] [image] [image] [image] [image]
> 
> This use two 18350 battery. One 144 LED strips. [image] [image] [image] [image]
> 
> This use 4 18350 battery. Two 144 LED strips.
> 
> [image] [image] [image] [image] [image] [image]

  can you help budy