MartinNohr/MagicImageWand #39: Some questions about the new version with TTGO part

ThomasKlee64, 2021-05-24:
Hello Martin,

I came across this project with great interest and still have a few questions for me as a beginner!
I have read that there are already previous versions of the project, among other things with the ESP32 WIFIVERSION 2HELTEC KIT 32. Now a different device is used. I'm not quite clear from the schematic. Here the ESP32 TTGO device is operated without 3.3V?

[https://github.com/MartinNohr/MagicImageWand/blob/main/TTGO%20Magic%20Image%20Wand%20Schematic.pdf](url)

And what's the deal with the wiring diagram remote. I apologize for my perhaps stupid questions. If you could help me here the start in this project something I would be grateful. 

[https://github.com/MartinNohr/MagicImageWand/blob/main/TTGO%20Magic%20Image%20Wand%20Remote.pdf](url)

THX

Thomas

zhaoyu3517, 2021-05-24:
TTGO display can use  5V. So you can input 5V or 3.3V or use lithium battery. But only one can be used, not at the same time.

ThomasKlee64, 2021-05-30:
Thanks for the answer

closed 2021-05-30 by ThomasKlee64

MartinNohr, 2021-08-13:
I'm not sure I answered this completely yet.
TTGO has its own 3.3 V regulator, so you can feed it with 5V and it works just fine. The 5V is necessary for the WS2812 LED strips so you have to have 5V around anyway. The data line to the LEDS seems to work with the 3.3V control lines, but it is out of spec, that is why I added level shifters to the PCB I designed. Of course the PCB also makes all the wiring easy since the rotary dial switch is on the PCB. My latest PCB even has a 5V regulator on the board. I created a new design that now uses a pair of 18650 batteries in the handle. Works much better and is much easier to build.
The remove schematic is for another device that is used to remotely trigger the camera. I haven't finished the software for that yet, but I will get it done eventually. It will automatically open the camera shutter when the wand display is started. You don't need an assistant or a remote control for the camera!