MartinNohr/MagicImageWand #42: MagicImageWand with larger 3 or 4 inches Display

ThomasKlee64, 2021-05-30:
Hello Martin,

I have a question. I have rebuilt the MagicImageWand with a lot of friends and I am excited about it. I would like to try to build the whole thing with a larger TFT display 3 or 4 inches and an ESP32 NodeMCU Module WLAN WiFi Dev Kit C Development Board. Is it possible to adapt the code?

Many greetings

Thomas

MartinNohr, 2021-06-02:
Yes, but I am not sure how much work. If the display works with the tft_espi library it should be pretty easy to do. You will probably have to change some of the GPIO pins for the SD card and the rotary switch.
I am in the process of porting the code to the TTGO T4 which has a 2 1/4 inch display. It is supported by Bodmer's TFT library so it is very easy to do. The board also has an SD reader.

MartinNohr, 2021-06-05:
I reopened this as I am working on the TTGO T4 with a larger display. I'll post information here when it is ready.

ThomasKlee64, 2021-06-05:
Great news, am looking forward to the first results.

MartinNohr, 2021-07-08:
Coming soon on this project. I just had to get some other projects finished. Maybe by September I'll have the code working.

DaveStu, 2021-11-16:
Hi Martin, hope you're well. It's been a while my friend. I'm wondering as you're porting the code to a larger display board, does the current uploaded code uploaded still work with the  original TTGO? 
I've tried compiling it and it breaks with 'gfxglyh does not name a type. in fonts.h

I'm wondering if this is the newer code for TTGO T4 or are they interchangeable as they use the same microcontroller?
Thanks

MartinNohr, 2021-11-17:
I haven't seen that error. I haven't ported the code to any larger display yet and I don't think I changed fonts.h. Weird.
It should still work with the original TTGO, that's what I'm using.

DaveStu, 2021-11-17:
Hey Martin, I did assume that as it it would have been documnted and/or new project started. 
So, I did a refresh of the Arduino program (always a good starting point). Followed the guide, set it all up, added and updated libraries, the ''gfxglyh' error has gone, though now it breaks at 'Error compiling for board ESP32 Dev Module'.
Im using yout latest code V1.15.

I'm on Arduino 1.8.16 with latest esp32 library V2.01. Then I tried V.2.00, V1.06 and V1.04 (Shown on your screenshots) and have the same problem. Failing that, perhaps an older version of Arduino, but would'nt make any logical sense?

On a different realted problem. I still have code V1.04 installed on the board so have been testing that whilst I get other issue figured. As I want to use two strips, I was trying to follow your section on 'Led Wirining Mode' and want to use mode 2 to create a "no black gap". 
Now from my basic understanding, the strips are still wired the same as nomal being directional, the strips themselves are turned around/reversed?
I thought I was right, as when I execute 'cylon eye' pattern, STRIP1 is now reversed, but STRIP2 does not light up.
After a short delay, the sequence returns to STRIP1, the delay being that STRIP2 should be lighting up.
Now, do I actually need to wire them differently? (I was slightly confused by document) or is their a bug in code V1.04?
Of course as per instructions, no electrical contact in the middle is made.

The Idea is, that I carry two strips in half in a case then connect them up on site, because they are long.

DaveStu, 2021-11-17:
My error log

In file included from D:\ARDUINO\MagicImageWand\MagicImageWand.h:34,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/sdfat.h:438:2: warning: #warning File not defined because __has__include(FS.h) [-Wcpp]
 #warning File not defined because __has__include(FS.h)
  ^~~~~~~
In file included from C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_rmt_esp32.h:116,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:8,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/platforms.h:36,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.h:58,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
C:\Users\SUPERMAN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from C:\Users\SUPERMAN\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:84,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:65,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.h:61,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
C:\Users\SUPERMAN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\FS\src/FS.h:30: warning: "FILE_READ" redefined
 #define FILE_READ       "r"
 
In file included from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatFile.h:821,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatVolume.h:27,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatLib.h:27,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/sdfat.h:33,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.h:34,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
c:\users\superman\documents\arduino\libraries\sdfat\src\common\arduinofiles.h:31: note: this is the location of the previous definition
 #define FILE_READ O_RDONLY
 
In file included from C:\Users\SUPERMAN\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:84,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:65,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.h:61,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
C:\Users\SUPERMAN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\FS\src/FS.h:31: warning: "FILE_WRITE" redefined
 #define FILE_WRITE      "w"
 
In file included from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatFile.h:821,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatVolume.h:27,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatLib.h:27,
                 from C:\Users\SUPERMAN\Documents\Arduino\libraries\SdFat\src/sdfat.h:33,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.h:34,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
c:\users\superman\documents\arduino\libraries\sdfat\src\common\arduinofiles.h:35: note: this is the location of the previous definition
 #define FILE_WRITE (O_RDWR | O_CREAT | O_AT_END)
 
D:\ARDUINO\MagicImageWand\MagicImageWand.ino:3454:11: warning: multi-character character constant [-Wmultichar]
   a1[0] = '0x7e';
           ^~~~~~
D:\ARDUINO\MagicImageWand\MagicImageWand.ino:3456:11: warning: multi-character character constant [-Wmultichar]
   b1[0] = '0x7e';
           ^~~~~~
In file included from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
D:\ARDUINO\MagicImageWand\MagicImageWand.h:3:19: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 char* myVersion = "1.29";
                   ^~~~~~
In file included from D:\ARDUINO\MagicImageWand\MagicImageWand.h:58,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/FastLED.h:65,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.h:58,
                 from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
C:\Users\SUPERMAN\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from D:\ARDUINO\MagicImageWand\MagicImageWand.ino:7:
[... 234 lines cut ...]
D:\ARDUINO\MagicImageWand\MagicImageWand.h:1041:1: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
[... cut ...]
C:\Users\SUPERMAN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\WebServer\src/WebServer.h:141:5: error: call to 'StreamFile<FsBaseFile, long long unsigned int>::name' declared with attribute error: use getName(name, size)
     _streamFileCore(file.size(), file.name(), contentType);
     ^~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\SUPERMAN\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.