MartinNohr/MagicImageWand #45: What is the function of the inputs for the frame sensor and the light sensor?

ThomasKlee64, 2021-06-30:
Hello, i have a question. What is the function of the inputs for the frame sensor and the light sensor? What can be done with this?

[image]

THX

Thomas

MartinNohr, 2021-07-08:
The frame sensor is for a wheel with an encoder that you attach to the bottom of the wand. You can roll it and it will advance the image to the next frame after every N pulses. This make for very accurate image creation without having to watch how fast you're walking. Unfortunately I haven't finished the design of the wheel/encoder yet. The code is working though, I have tested it.
The Light Sensor is an idea I had to control the display brightness and possible also the wand brightness depending on the ambient light. There is no code yet, it is something I plan to experiment with soon. I added it to the schematic just in case it turned out to be useful. You can never have too many features. :-)
The light sensor could also be an infrared detector which you could use to start the wand by using an infrared light pointed at the wand. How useful??? I don't know, It was cheap to add to the PCB so I thought I would put it on there.
I am also almost finished with my latest PCB design. It now has a 2Amp power supply on the same PCB. This makes the whole thing easier to build, not as many wires to hook up or parts to get. It'll probably be September/October before I have those PCB's though.

ThomasKlee64, 2021-07-27:
I tried to test the function with the frame sensor yesterday, unfortunately without success. I could not find a function with the GPIO pin 32 in the program code. Is this part of the program still missing?

ThomasKlee64, 2021-08-03:
> The frame sensor is for a wheel with an encoder that you attach to the bottom of the wand. You can roll it and it will advance the image to the next frame after every N pulses. This make for very [...]

closed 2021-08-03 by ThomasKlee64