MartinNohr/MagicImageWand #49: image preview option glitch

DaveStu, 2021-12-27:
I've just flashed v2.03 and the option to scroll and preview images has started to glitch for me.
In the previous version (v2.02 I believe) worked perfectly.

When I scroll to preview next image, it flashes and reverts back to previous one.

MartinNohr, 2021-12-29:
Try 2.09. There were some problems I fixed.

DaveStu, 2021-12-30:
Ok, will try it next. Just waiting for a SDCARD board, as I've damaged the pads on my first one and cannot find any traces to connect to.

DaveStu, 2022-01-06:
Well new SDCARD board arrived, connected it up and all was well with the prevous code.
Tried to then flash your latest code and the TTGO says "ESP is not in flash boot mode" with ESP home flasher.
I've tried different cables checking connections for any shorts, now I'm stuck.
I'm sure its a driver issue, so tried to clear them but to no avail. Frustration doesnt't even describe things.

DaveStu, 2022-01-24:
Version 2.21 is a night and day difference in dial interaction, I find it glitch free and wonderfully responsive.
Kudos to the effort Martin.

MartinNohr, 2022-01-24:
It took me all day but I finally got the dial working right! Time well spent though, it's a delight to use now. Glad it's working for you too.

DaveStu, 2022-01-26:
I see, glad no one asked me to code it then...eternity. It's definately a delight to use. It's great for all.

MartinNohr, 2022-01-26:
Dealing with mechanical switches that bounce is always fun! I should have debounced it in hardware on my PCB, but that also have meant more parts for the hand-built people, so I decided it was 'easy' to fix in software. :-)

MartinNohr, 2022-03-05:
I'll close this one, it seems to be working very well now.

closed 2022-03-05 by MartinNohr