MartinNohr/MagicImageWand #50: PCB and components

DaveStu, 2021-12-27:
Hi Martin, Hope your chrostmans was good. I was wondering If one could get the gerber pcb files, and also if BOM + CPL lists are available, so I could get the PCB printed by JLCPCB?

My handmade board is too fragile to be used in the field and requires a more professional, robust board

Thank you

tobias966, 2021-12-29:
Hi Dave,

you can order the PCB there

https://www.pcbway.com/project/shareproject/Magic_Image_Wand_V2_with_PowerSupply.html

Regards
Tobias

MartinNohr, 2021-12-29:
That is the bare pcb of course. PCBWAY won’t let you order single populated PCB’s. I think you have to order 10 at a time, which is what I do, well, actually I order 20 sometimes.

I have a new design of the PCB, but I haven’t got the samples back yet so I don’t want to make it public until I have a chance to test it. No major changes to it though. I moved the power supply components to make it easier to put heat sinks on and I added another connector that I need for my remote camera control and my 3D printer filament weigh scale. I use that PCB for multiple projects.

 

https://www.pcbway.com/project/shareproject/Magic_Image_Wand_V2_with_PowerSupply.html

MartinNohr, 2021-12-29:
https://usa.banggood.com/LILYGO-TTGO-T-Watcher-BTC-Ticker-ESP32-For-Bitcoin-Price-Program-4M-SPI-Flash-Psram-LCD-Display-Module-p-1345292.html?cur_warehouse=CN&rmmds=search
Here is another one that would be nice. I have actually ported the code to run on a card with this display and a touchscreen. I didn't really like the touch screen, it always needed two hands, but perhaps a version with the large screen and my rotary dial would be good. I'll work on that some more next year.

MartinNohr, 2021-12-29:
Here are the gerber, BOM, and pick files for the onboard power supply PCB.
https://github.com/MartinNohr/MagicImageWand/tree/main/MIWNPS1_2

MartinNohr, 2021-12-29:
This board is available and would be pretty easy to move the code to.
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20211229110930&SearchText=ttgo+t4
It has a lovely 320x240 display. It does not have touch screen, but I'm not convinced that is really useful. It would be easy to connect a power supply and the rotary dial and the LED outputs to here. A PCB could designed that this card can be plugged onto. Notice that is has the micro SD reader already there.

DaveStu, 2021-12-30:
OK, I'll perhaps wait for your updated PCB design, though perhaps pointless for me as you use the PCB in other projects too. 
As you've said, PCBWAY wont let you order single populated or a few, So I thought I'd try JLCPCB. 
Thanks for the BOM list.

That board from Banggood is the same one from Aliexpress, someone has just designed a case for first one.
Seems decent in screen size though.

MartinNohr, 2021-12-30:
The new pcb has a nicer layout for the power supply parts, otherwise it hasn't changed much. I have 3 of those ttgo t4 boards. Lovely displays. I've been planning to port the code to it. Sent from my Verizon, Samsung Galaxy smartphone

MartinNohr, 2021-12-30:
I think that JLCPCB also has some minimum limit of assembled boards.

I always have some here, but shipping costs from US to the rest of the world are very high. I sell a PCB for $35 and it often costs $45 to ship it with the post office. I don’t know any way to make it cheaper.

DaveStu, 2022-01-01:
Hi Martin, what format or tool did you use to export the BOM and P&P list, as I can only view the CSV and trying to save leaves a file with data thats jumbled and not tabulated? 

Indeed, JLCPCB have a minimum of 5 boards. PCBS alone woud cost around $10 shipped to UK, but components more.
Me buying depends on their price

MartinNohr, 2022-01-01:
I just exported those files from EasyEDA. They are a standard CSV (text with columne) type format. I can read them with either Excel or NotePad++. Just make sure if you make any changes in Excel to save in CSV or it will be converted to Excel and then it will not be in CSV anymore and in fact might appear jumbled! I can sell you the populated/tested PCB for $35, but the shipping is a killer. $40.85US. If you know anybody else that wants one I can fit at least 3 in the small flat rate box. Maybe 4, but I've never put more than 3 in there.

MartinNohr, 2022-01-01:
Interesting, I was just looking on the USPS.com site. It appears they have a 1st class package service to UK for $16.25. I haven't used it before, it might be slow, but the price is certainly interesting.

MartinNohr, 2022-01-01:
Re the 1st class package to UK... Actually I have used it, just not to the UK, I sent some PCB's to Canada with that. It took 3 weeks to get to Edmonton, but the price was good. So that is an option that might be worth considering.

MartinNohr, 2022-01-03:
I have been looking at the TTGO T4 some more. I have a couple of them here. It would be easy to port the code to, but looking around the internet I am worried about the availability of this board. It is only sold a couple of places so unless that changes this card my not be available in the long run.
A more interesting idea may be to use the M5. I have already ported the code to it. It would be easy enough to make a little PCB that has the rotary switch (which I still prefer for ease of use) and the level translator drivers for the LED's. The M5 has a lovely display. You remove the bottom plate and then it could plug directly into another PCB with the dial and drivers and perhaps even a nice powerful 5V regulator.

DaveStu, 2022-01-11:
Indeee, that TTGO T4 is very limited in availability, I noticed that too. Are you refering to the M5stack core2 esp32?

MartinNohr, 2022-02-09:
Yes, that is the one, the grey one that has physical buttons. The white one has touch screen buttons and I don't like them as much, too hard to use in my opinion. I have an early version of the code working on it, it's not up to date but it does show how it works with a redesigned menu system.