MartinNohr/MagicImageWand #51: reboot necessary

tobias966, 2021-12-29:
Dear Martin,

today I used the first time led stripes on the MIW. I have the software version 2.07 on it. When it boots and the version appears on the ttgo also the led on the stripe (I have only on mounted) runs with rainbow color from start to end and back. So for me everything is right. The point is, that I can´t run the images on the led. The preview and all other menues etc. are working, but the led strip doesn´t react. Only after reboot, with the reboot function in the menu it is working. 
It doesn´t matter if I put the power with USB C directly on TTGO or with the battery power over the PCB. Batterypower I use 2x 18650cells in line.

With the batterypower I have another phenomen. The TTGO doesn´t start. I can meassure 5V on pin1 and 3,3V on pin12 also I have 5V on D1 top or on PCB points LED3 and LED4. So for me it looks fine. But I have to press, the small reset button on the side. I don´t have added the frame sensor or any other things on the pcb, just one led stripe at LED1, could this be the reason?

Hope you can support.
Tobias

MartinNohr, 2021-12-29:
Not sure. I run 1 LED strip most of the time and it works fine. Have you tried any older versions of the software? I might have broken something with all the changes I made. You can also try “factory reset”. Hold the dial button down while applying power.

If the rainbow pulses when you turn it on then it indicates that the LED strip is working correctly. And if the preview of image files work then the SD card reading must also be working.

Do any of the built-in patterns work? Does the lightbar function work?

The frame sensor isn’t used unless you have enabled it in the menus. That’s why I suggested to do a factory reset operation.

Are you using the WS2812B with 144 LED’s?

Does the progress bar move when you try to show an image? Or what exactly does happen?

MartinNohr, 2021-12-29:
One more question: So it works when powered from USB without batteries?

MartinNohr, 2021-12-29:
I am wondering, did you set the controllers to 2 and then display an image file of 144 characters? That would give the exact symptoms you are seeing. The rainbow would show on start, and the preview would work. The actual LED display would fail because LED1 is the pixels above 144, which aren’t there if the image is 144. You could enable the image doubling command to make it work.

 

About the restarting though I have no idea, I have never seen that. You might try a different TTGO.

tobias966, 2021-12-29:
Dear Martin,

I checked no several things. The behavour is always similar. It doesn´t matter if power from battery or usb without batteries. I also checked another led strip both with 144led , WS2812B. The controller set is 1. Also factory reset doesn´t help.
During first start I see rainbow over the complete length of the strip. Preview is also working. But when I want to show the image, only the first 6 leds are working. No difference between built in pattern and images from sd card. When I change to light bar all colors exept green are only working with the first 6 leds. When I increase HSV to 10 or more the 7th led is lightning in green. This stops after HUE150 then only the first 6 leds are working. After reboot everything is working.
In the next days I will build up another pcb, will check what happend after. Because this is the only thing, which I don´t changed up to now.

MartinNohr, 2021-12-29:
It is very strange that it shows the rainbow on startup. That means it can write to the LED's. Have you tried any of the built-in patterns?

tobias966, 2021-12-29:
yes built in patterns have similar behavour. Let me try a new pcb in the next days. Honestly I can not explain which influence it should have, but we will see, also I ordered a new ttgo hopefully it will come in the next days.

MartinNohr, 2021-12-30:
The only time I have seen anything like that is when I had long LED wires and my original PCB had the wrong kind of driver chip on it. I looked at the LED data line and saw that it had lots of "ringing" and noise. If you have an oscilloscope you could take a look at the data and power lines on the LED.

tobias966, 2021-12-30:
I have no oscilloscope but after 10th I will check it in my company, hope they will support me. With driver chip you mean the U1? This is from nexperia similar like in your bom. Should I add additional resistors or the 100k resistor on the pcb are enough?

tobias966, 2022-01-05:
I meassured the signal with the oscilloscop. Directly on the pin7 (2LED1) at the ttgo pcb. Also I meassured the signal on the pcb with power supply and on my experience board, where I have only connected the rotary button and nothing else, power supply was with usb direct on ttgo pcb.
Softwareversion was 2.10 and the file which I started was Mario1 which I got from you when I bought the imagepainter from you (last year).
signal after start
https://user-images.githubusercontent.com/73199898/148271391-4cf865b6-5b59-4c09-ae7b-56d9b843cb2f.mp4
signal after reboot
https://user-images.githubusercontent.com/73199898/148271409-08d41c99-07a3-4029-8b26-8b8dd64d998b.mp4

I checked also another ttgo with newest software version 2.16 but with this version reboot doesn´t help. The led are not working, similar behavior as with 2.10 before the reboot.

Is there any possiblity to find the software version 2.10 because I don´t want to change the ttgo with 2.10 to 2.16 to check if it is also not working anymore, but then I have no possibility to lightpaint.

During the start of the ttgo when the rainbow led comes, the signal is also not complitly the same between first start and reboot start.

All measurements I have done without led stripe connected.
Hope this helps you to support me, to find out the root cause.

tobias966, 2022-01-05:
https://user-images.githubusercontent.com/73199898/148274389-4ed81d8f-54f6-4fa3-b9a3-188acb8fcf8d.mp4

First movie is not working, so again, signal after start

tobias966, 2022-01-05:
> 

https://user-images.githubusercontent.com/73199898/148274611-941cd9f6-62b5-4d17-9fb6-508e51d32db7.mp4

tobias966, 2022-01-05:
https://user-images.githubusercontent.com/73199898/148274647-f05e47af-0a04-4cbf-8372-2db61a75fb67.mp4

tobias966, 2022-01-05:
https://user-images.githubusercontent.com/73199898/148275254-7bd29bdb-9acb-47b7-baa8-e50905272406.mp4

tobias966, 2022-01-06:
I found software version 1.28 on my hardware. With this software it works without reboot function. So I think there is something different in software code between 1.28 (works without reboot), 2.10 (works with reboot) and 2.16 (doesn´t work, no matter if rebooted or not). 
If you want I can make also a small video with output signal of version 1.28

MartinNohr, 2022-02-18:
The 100 ohm resistor is to minimize the ringing on the lines. The signals on the 'scope look ok. I'm puzzled and confused!
The LED strip does need to be connected to really see the signals, put the probes right on the LED connector. Check the strip voltage also, it should be very close to 5V.

MartinNohr, 2022-02-18:
What brightness level are you using? USB power won't run them very bright.

tobias966, 2022-02-20:
the brightness level is 25. Changing to 5 makes no difference.

tobias966, 2022-02-21:
check the strip voltage it is always 5V also no power drop (don´t see on oszi) during the start. Also no difference if you use external usb power or the pcb power
still have feeling to use some wrong libaries. Most influence i found with esp32 board controler. Can you tell which version you use for the board controller?

MartinNohr, 2022-02-21:
Try the binary file that I compiled. It is here on GitHub.
Here are my settings from VS2022.
[image]
How do I tell which board controller I'm using?

tobias966, 2022-02-22:
thanks I will check. Up to now I´m using arduiono IDÈ. I will try also VS2022 and see how find the info in VS

tobias966, 2022-02-22:
I tried the binary file (sorry didn´t know that this is available). With this it works, so no reboot necessary. Thanks to give this idea.

MartinNohr, 2022-04-24:
Since the binary version works, it must be some kind of library issue.