MartinNohr/MagicImageWand #52: Dial problem with on-board power supply PCB

MartinNohr, 2021-12-29:
I noticed a problem with the dial incorrectly reading rotations and clicks when running from batteries. Rotating the dial fast will cause it to start reading phantom clicks and other things. I thought it was a software issue, but it never happens when running from USB power. I am investigating noise on the switch lines from the power supply. It might be necessary to add some R or C elements to the board to fix this. I will post the solution here when I have it.

DaveStu, 2021-12-30:
I've concluded something similar and from investigations, it seems the LM2596 buck converter board I'm using from the original MIW parts list could be the culprit. Apparently the ripples could be quite high and datasheet suggest the output capacitor needs changing depending on voltage requirements.
Not sure how it setup on your PCB though.

MartinNohr, 2021-12-30:
My 'scope shows the 5V is clean and so are the dial and switch lines. So I don't know exactly what is going on. It does seem to only happen after previewing the image, but it also only happens when running on battery power. This is proving to be a hard one to find. I have made some interrupt code changes that seem to make a little bit better but it still happens.

DaveStu, 2022-01-01:
I see, It's just what I found in forums. I haven't got a scope to analyse so I'll assume it's not the buck converter.

MartinNohr, 2022-01-01:
Happy New Year to you and all around you.
It is certainly odd that the quickly rotation dial can go weird but only on battery power. It certainly smells like the power supply, but on my 'scope it looks fine. I wish it would show the problem when running on USB power, then it would be trivial to find and fix. Oh well, maybe this year I'll figure it out... :-)

DaveStu, 2022-01-08:
code v2.15 seems much better on battery power than before.

MartinNohr, 2022-01-09:
I have now seen the same problem using USB power, not as often, but it does happen. So I can now see what happened. The click dial input gpio always reads 0, but the port is actually high. Very strange!

DaveStu, 2022-01-19:
I wonder if its different for you, because I'm not using your PCB but more simpler connections and less components?
Although it's not glitche free, as you've experienced.
I read a long while ago about pullup resistors and rotary encoders, that the esp32 dont neccesarily need external ones, as you can activate internal ones? I made a note about the code being enc_internal_pullup and setting this to -1 would activate them. I'm not a coder, Is that something you've done or we need?

MartinNohr, 2022-01-19:
I am using the internal pullups, although they are not available on ports 34-39 (if I remember correctly). It is handled in code, if you look in rotarydialbutton.h in the begin() function you will see where the direction and the pullups are enabled.

I have fixed it now, I monitored the switch with my ‘scope and found that while showing the preview and rotation the dial fast the push button input turns into an output. I am thinking this might be some quirk in the esp32 TTGO hardware. Anyway, I worked around it with a little software and by moving the push button contact to a different GPIO pin. I used 35 which is input only!

BTW, I have seen the problem even when running with USB power, so I don’t think the onboard power supply is really the issue. Although, it is harder to do with USB power. I looked at all the voltages with my ‘scope and everything looks fine. I am suspicious of the SD code which is heavily used when previewing the image. I am using a very old version of the SD code. The newer versions are not compatible with the old one so I haven’t upgraded. I suppose I should do that. It might fix the problem.

Martin

DaveStu, 2022-01-19:
I had a look, pardon my ignorance, you are right about the pullup code. It's not something I ever work on, just read about.
I am certainly learning though.

Thats interesting about the SD code, again, you are the master at this.

MartinNohr, 2022-01-23:
It appears to actually have been a problem in my interrupt handling code, I'm not sure exactly what, but version 2.21 seems to be working perfectly on battery or USB power. I rewrote the dial interrupt handling.

closed 2022-01-23 by MartinNohr