MartinNohr/MagicImageWand #53: preview of 288 pixel bitmaps

MartinNohr, 2022-01-26:
Preview of 288 pixel bitmaps is broken. Scrolling causes very strange effects.

DaveStu, 2022-01-26:
Indeed it does. Mine seems to duplicate about 20-25% of the image on the right hand side.
Not a deal breaker and certainly usable.

MartinNohr, 2022-01-26:
It's a "little" math error! I'll fix it soon. ;-)

DaveStu, 2022-01-27:
No rush, the main culprit was the dial interaction, that issue did get annoying

MartinNohr, 2022-01-27:
I've almost got it working, just a little arithmetic problem when scaling down to 144 from 288. I should have it fixed in the next couple of days.

MartinNohr, 2022-01-27:
Fixed. It works for me now! Let me know if any problems are found. I'll leave this issue open for a few days.

MartinNohr, 2022-01-31:
Since it seems to be ok now, I'll close this thread.

closed 2022-01-31 by MartinNohr