MartinNohr/MagicImageWand #54: Request to make builtins/SD images lists static

DaveStu, 2022-01-26:
I was wondering if its possible to, when scrolling images on sdcard or built-ins, that the cursor bar moves up/down instead of the list, similar to how the main menu options are driven?  Of course when the list is longer, it simply scrolls up, again like main menu. 
Perhaps this could be added as an parameter to choose or fixed globally?
This will also keep it consistent with the main menu as for me it becomes slightly confusing and less flowing.
Unless theres a reason for it?

MartinNohr, 2022-01-26:
There isn't any good reason for it working that way. I have often thought of changing it, I just haven't done it yet. The only thing I can say is that it leaves the current filename on the top of the display, same place as it is when you run the file. I would need to do a little housekeeping to remember the offset after running the file, but it is doable. There is an option to not display any files below the current selection, then it only shows the current one. This is mostly left over from the days when I used a single line display. I'll give it some thought. I'm a bit busy right now, so it will be a couple of weeks.

MartinNohr, 2022-01-27:
One thing about leaving it at the top is that the same line is displayed when you run the file. This would now suddenly jump to different text during running when the cursor was not at the top. It would then return after file finishes. Is that OK?
I think if I add this it will be an option so the user can choose.

DaveStu, 2022-02-15:
Hi Martin, sorry have been busy with work and took a much needed holiday to get some cultural stimulation and mental recharge.
I'm not exactly sure what you mean, but it's not a deal breaker if not. If you do add it as an option, so be it.
In my mind I'm visulising it different, but you're the coder, you understand it better.

MartinNohr, 2022-02-18:
Check out version 2.24 that I just uploaded. I added an option that allows the file menu to act like the other menus. I made it the default. It's a bit quirky about displaying the folder path, I may change that, but let's try this first. Let me know if it works and how to improve it. I will say I like the consistency better as well!

DaveStu, 2022-02-21:
Excellent Martin. For me at least, I now find this more cohesive. Also, when I'm scrolling/searching for patterns, I can continue to glance five at a time, and not being off screen.
I haven't noticed anything about the folder path. May I suggest, perhaps it would be easier if at the top of the list it states something like "Previous folder" insead of current folder name? For me I understand the syntax, perhaps for others it's confusing?

MartinNohr, 2022-02-21:
I made some improvements in 2.26. The display is more consistent with folders being shown or not.
Regarding your comment on the folder on the top of the path: I get your point, but I think it is more consistent to show the current folder. I do regret using '<', though, it should probably have been '..'. I just didn't want to use 2 characters. On the other hand, perhaps it would be better to say something like "Previous Folder". But this raises the question, what if you're down more than one level? Then what should I put in there? I guess "Previous Folder" would work, but now you've lost the information about being down 2 or more levels. Needs more thought.

MartinNohr, 2022-02-21:
Another thought: when the cursor hits the bottom of the screen it scrolls, perhaps it should scroll when it hits the next to bottom line so that the next file is always visible?
User interface design is never easy!

DaveStu, 2022-03-05:
sorry Martin, have been busy then also the Russia Ukraine crisis turned to a war, as you will know.
I had an ex colleague who I worked with, who moved back to Ukraine and has been desperately trying to get out the counrty with her family, so I have been helping from UK with correspondents and communication. Thank fully they made it to Poland for now. Hope you have been well.

Yes you last mentioned idea sounds good and you're certainly right, user interface is difficult and also GUI design.
Well, I think your efforts have made it rather slick thus far.

MartinNohr, 2022-04-05:
Yeah, the Ukraine situation is very bad. I'm glad your friends made it out. Sad that so many haven't made it out. I have friends from Ukraine also.