MartinNohr/MagicImageWand #57: Save state when updating code

DaveStu, 2022-02-15:
Is it possible when an update is uploaded, that the previous auto/manual saved settings can be loaded, either automaticslly or manually?

Each time time I set my display up the way I like/need it and other configertions, after a new code is uploaded all previous settings are gone.  I do find this part annoying. I assume the saved settings are stored on internal flash and gets wiped at each new upload? If so, is it possible to save settings to SDCARD, thus allowing easy application of settings UNLESS features change considerably that the saved state causes corruption  to non-existent or changed menus options? I'm not sure on the logical way to tackle this, yet.

MartinNohr, 2022-02-18:
Just to clarify.
- Save your settings
- Load new firmwary
- Load your settings
You should then make sure they are correct, including any new ones that have been added, and then save again.

DaveStu, 2022-02-21:
Thanks for the insight and clarification. My step was to first turn on AUTOLOAD SETTINGS and I think, it would do an autosave of current state, thus overwriting my previous saved settings?
As, after doing that and THEN trying to load previous saved settings, there was nothing.
I will need to do as you said, THEN turn on AUTOLOAD SETTINGS.
I'll try for next version of code, as I've already applied your latest, before seeing this message. Thanks

MartinNohr, 2022-02-21:
I didn't think it was a good idea to do a save when autoload was turned on, that would overwrite an existing settings file. I suppose I could have designed it to do an autosave if no saved file was saved, but I hadn't thought of that at the time. I think about changing that, I don't think it will cause any problems. However, turning autosave on will still use an existing settings file. I suppose I could ask if the current settings should be save. But then again, hmmm, things get complicated!