MartinNohr/MagicImageWand #59: TTGO T4 Version

MartinNohr, 2022-07-15:
There is a nice esp32 board available with a 2.2" 240x320 display called the TTGO T4. I have been working on porting the code to this board and it is almost working. It even has an IMU chip that can be used to show whether the wand is level or not. The display is beautiful. I will post more information here as I finish the development of the code.

MartinNohr, 2022-07-17:
It is now working reasonably well, lots of testing is needed of course, and some button presses might need to be redefined to be more useful.

DaveStu, 2022-07-20:
Hi Martin, hope you're doing well, its been a while. Seems like you're full steam ahead with the T4 port of the code, excellent.
I've been enjoying the use of my esp32 based version as image painter of course but also impromptu light bar using it to light various colours as accents lights in photography. 

I have a couple of old android phones, which are not worth much in resale, so use them as wifi only devices for testing apps or even as a light usage NAS, or smart controller for lights and switches etc.
From that I had a crazy idea recently, although we dont need to implement it.

Imagine using a spare android phone as your interface to the esp boards, that giant screen in comparison
The code would obviously reside on the esp, but perhaps the phone is the commander and dictates code too.
Ideally hardwired so connectected via its USB port to the esp. 
I've been playing around with WLED and their app, hence the crazy idea I had.
Anyway, its not really required, I'm just a crazy thinker

MartinNohr, 2022-07-27:
It is working on a T4, feel free to try if you have one. It does need a little refinement on the button use. It uses the 3 buttons on the T4 instead of the rotary dial. I could hook up the dial but there aren't very many pins available on the T4 connector. It's very nice having that large screen.
I actually did have a ble phone app working with the MIW. We commented the code out because the guy in germany added the rudimentary web interface. BLE and WiFi can't be used at the same time.
However, I could get busy and refine the web interface to handle everything, it only does a few things right now. Using a web interface means a computer or a phone could be used to control the MIW. I'll start thinking about that again. Right now it does allow up and download of bmp files. It can adjust a few of the settings, I need to add some code to handle all the settings, then I think it might be useful. The display is certainly better. I think I already have it displaying the list of bmp files, but I haven't used it in a while.
I suppose you could hardwire it though the USB port, they can be used for serial data but I am thinking that the web interface might still be better.
Good to hear from you again, very few comments these days!
Oh, and your idea isn't entirely crazy!

tobias966, 2022-08-17:
Hi Martin, yesterday my T4 arrived and first look is it works perfectly after I understood working with the 3 buttons. At the weekend I will also use it with the led bars and give you a feedback. Because for me the rotary dial is easier to use at the moment, can I solder this also on the T4. Honestly I don´t check if the I/O´s are available.

MartinNohr, 2022-08-18:
I'll have to define some pins for the rotary switch and enable that in the code. There aren't very many pins available on the T4. I do agree, the rotary dial is much easier to use. I'll let you know when I figure out which pins we can use. Some of them are reserved, even though they are on the connector. Poor design.

DaveStu, 2022-09-02:
I haven't jumped on the T4 train, I'm not sure I need it unless there is a big leap in additional functionality plus a cost issue for board and 3D printing 
For me, the T1 screen size is ok, for now. I'll still be checking on progress and any input I can add.

I'm also currently trying to design/invent another gadget, well two. I'd love to be able to get out of this rat race and be healthy and financially stable by making it big and help a few struggling people along the way.

MartinNohr, 2022-09-06:
The only real advantage I can see for the T4 is the larger screen. Now there is a new TTGO T-DisplayS3 that is about the same size but with a slightly larger screen. I've ordered a couple of them to experiment with. Unfortunately it might not work in the existing PCB because while it has the same number of pins, some of them have changed positions. I might be able to work around it with software since the power and ground pins are the same. I'll post a comment when I know for sure.

manussn, 2022-11-03:
Can you help me? I was trying to configure the TTGO T4 wifi through the Arduino IDE and the screen turned off and did not turn on again. I don't know if I reset the EPROM. I'm desperate.

manussn, 2022-11-03:
@MartinNohr  this message was in the prompt.

**Leaving...
Hard resetting via RTS pin...**

MartinNohr, 2022-11-04:
Was this just an update or was it the first time the MIW software was loaded?
The "hard resetting via RTS" is the normal message for rebooting the ESP32.
Did you set the T4 correctly in both .h files? I think I put instructions in the docx file. If this isn't done then the display will not work at all. I suspect that you didn't change the two .h files, one in the MIW code and the other in the display driver library.