MartinNohr/MagicImageWand #60: no access to sd card

tobias966, 2022-08-30:
Dear Martin,
today I wanted to try your new software files but I got always same error. After the startscreen it comes the message:
creating : /macro.json and then in red letters failed to open : /macro.json
And then I have only the built in images can´t switch to sd card.
This failure I have for software version 2.41, 2.40, 2.39 but not for 2.36
2.37 and 2.38 I don´t have tested.
I tried it by compiling myself with Adruino IDE but also I uploaded your bin file which is in the folder release.

I hope you can help me to find the issue.
Regards
Tobias

MartinNohr, 2022-08-31:
Due to some issues with exfat I decided to switch back to the standard SD library. So I’m guessing you have an SD card larger than 32GB or it has been formatted with exFat. True?

You can compile with the USE_STANDARD_SD set to 0 and see if that works.

Martin

tobias966, 2022-08-31:
Thanks for support it was a 64GB card. I changed to a 2GB and it works.

closed 2022-08-31 by tobias966

MartinNohr, 2022-09-01:
In the .h file, you can set USE_STANDARD_SD to 0, that will use the exfat library and cards >32GB will be supported. I just don’t know for sure if the web server page is still working. I will test that eventually though!