MartinNohr/MagicImageWand #61: Text colour not saving, cannot exit menu v2.42

DaveStu, 2022-09-02:
In latest update (v2.42) not a critical issue although I do use it to make reading in the dark easier.
When I go to options to change colour of text, I cannot save and exit as usual. I've tried buttons, doesn't work so I need to reboot. 
This is on TTGO T1.

MartinNohr, 2022-09-02:
Thanks for catching that one, I hadn't noticed it yet. I just fixed it (2.43), please try the latest and make sure it works for you also.
Did you notice I added a feature that will install the bin file from the SD card?

tobias966, 2022-09-02:
I tested it on my ttgo1 V2.43 is working well. Also the new feature with the bin file is great. Saves a  lot of time, because I don´t have to compile every time.

MartinNohr, 2022-09-04:
Glad you like the load from bin file. I even use it myself to install the latest on the several models that I have of the wand. I think it will be very useful to people that can’t or don’t want to compile.

Martin

closed 2022-09-06 by MartinNohr

DaveStu, 2022-10-06:
Hello, yes all fixed now. Since me last input from version 2.42 you've certainly been busy, version 2.75!
First time I used the webserver, very funky and intuitive to use, excellent work. You've certainly put in a lot of work and features already. Although it's not something I need yet, it's a fabulous addition. Though my friend did comment, allowing to upload images straight from the phone is fantastic!
Oh yes, I did see the bin file upload feature, thats very useful indeed. You sir are full of surprises.
Just curious, I wonder what happened to Dirk?

MartinNohr, 2022-10-06:
I like to keep busy! Glad you like it. I’m pretty happy with my design, although I still have long list of things I want to improve. Doing this kind of thing keeps my mind sharp, I believe so anyway. I’m a retire software engineer (I’m 72 now), and I used to design analog and digital hardware. After my MSc EE, I went to Bell Northern Research for a few years (middle 1970’s). They were the Bell labs of Canada and I learned a lot there.

About Dirk, I have no idea, he’s a smart guy and I have appreciated his input greatly in the past. Dirk suggested the web server interface originally, I was writing a phone app to do the remote control. When he noticed that BLE and the web server wouldn’t happily work together I abandoned the phone app work. The web server is more general since it works from any browser, not just a phone. He hasn’t posted anything in a long time. I never had his direct email address, if I did I would try and reach him. He does seem to have just disappeared. I’ll let you know if I ever hear from him.

So what are using the MIW for?

DaveStu, 2022-11-11:
"So what are using the MIW for? "
So far, I've testing/using it with some of the wedding couples, so create something different, thouugh that is a very occasion due to not having many booking. 
Then also, I've been struggling with mental health for many years and I try to create or involve in such project, some for simply pure enjoyment and others to be able to learn new things and also enjoy photography, a hobby I use for my ,mind state, this MIW has allowed me to experiment in a different way and different time of day.
Recently I have had interest with a fellow photography, who wants to calibrate on some projects, with me helping her to create some unique light painting patterns, so this is where it's going too..

I've been trying to rework the handle design on mine, currently being round I find it a weak point, to try and create a sturdier grip, like a grab handle to easy to hold with one hand, and perhaps move or duplicate the "execute"button to handle in a ergonomical way that can easily be pressed.
Also how I could design something, that quickly allows me to attach a wheel or two to the bottom of MIW, like chunky 4" pneumatic wheels, which also are height adjustable (perhaps like a tripod height adjusting clamp mechamism).
Perhaps even a 4 wheel system, like a skateboard but with smother softer air filled wheels.
I can then roll the MIW along the contours of the ground, especially if its curved, to create more accurate images, especially when trying to create short animations with bmp frames, when I need to repeat the same moveent many times. I came to the idea from videographers who use sliders or dollys. 
I originally did think of gyros but thats far too complicated.

MartinNohr, 2022-11-15:
I'm glad you are using the MIW in several ways that are helping you and making you more creative.
Have you looked at my handle design? I find it easy to hold and the button is right under my thumb so it is easy to use.
I do like the wheel idea, I've been planning to design something like that with an encoder to advance the frames. I'm also designing a new PCB that has a motion detector/orientation sensor onboard. This will also allow the frames to be advanced according to the wand motion. It will take a bit of calculus, but it isn't too complicated to do the math. The faster you move the fast the frames will change! But of course using a wheel also lets the wand follow the ground and might be better. There is already code that advances the frame with each pulse from an encoder wheel. I just need to design a wheel using an encoder.

DaveStu, 2022-11-17:
I have indeed seen your handle design, it's now sort of where I've landed, after experimenting. I would have used your files, but printing was/is too expensive especialyl for strong plastics. So, as I like making thinks anyway, as I've always enjoyed DIY, woodwork and mroe, I put something rudimentary together 
Ah I see, you've been working on a wheel idea too, excellent. I like the advancing frame idea, is that for automation?
As in, would the camera take the next image too automatically. Whats your thinking behind it on how to use it, I'm intrigued? 

[image]
[image]

MartinNohr, 2022-11-17:
I printed with PLA, it was plenty strong. Cost a couple of dollars in filament. But of course you have to have a 3D printer.
By "frame" I meant a column of the BMP image. As the wheel rotates the encoder pulses the input to the MIW. Each pulse makes the MIW move to the next column in the BMP image. It provides perfect column advancing that uses position instead of the "frame time" in the MIW. The code is already in there for it and I have tested it. It's an option in the image timing setting somewhere, I don't remember exactly what I called it.
I am working on a remote control for the camera. What I've done is used the wireless to fire the camera shutter when the button is pressed on the MIW to play the BMP file. That way you don't have to have someone running the camera. You can do it all by yourself!