MartinNohr/MagicImageWand #67: MagicImageWand wiki improvement

jbreizh, 2023-10-24:
Hello,
i have put your word document into a nicer github wiki. I try my best to keep all the information, but it was a lot of reformatting so i can have made a few mistake....
Anyway you can find this new wiki here : [https://github.com/jbreizh/MagicImageWand/wiki](https://github.com/jbreizh/MagicImageWand/wiki)

Github wiki are simply a git repo, but Github don't provide a way to make  a pull request for it, so a few explaination to use my wiki :

* make a local copy of your old wiki : git clone https://github.com/MartinNohr/MagicImageWand.wiki.git
* make a local copy of my new wiki git clone : git clone https://github.com/jbreizh/MagicImageWand.wiki.git

So you can :
* save a copy of your old wiki (just in case).
* delete all the file and directory from your old wiki in  (don't touch the .git directory)
* copy all the file and directory from my new wiki to your old wiki  (don't touch the .git directory)
* add, commit and push from your old wiki : 
```
git add .
git commit -m "new wiki !!!"
git push -u origin master
```

You can now edit your wiki as github repo. I have made every address relative (picture from /images and also in Home.md and _Sidebar.md) and try to keep a logical arrangement with your word doc.

Have a nice day

MartinNohr, 2024-01-21:
I just noticed this. I'll consider it, but I will probably keep the docx file around. I use it to document everything. I haven't worked on this project for a while. I am now porting the code to run on the TTGO T-Display S3. There will be a new PCB for it.

MartinNohr, 2024-01-21:
The wiki looks great but it is 23 versions behind the current one. I made numerous changes late last year.