MartinNohr/MagicImageWand #75: Webserver compilation error

NashMerwolf, 2024-09-22:
Hi,
trying to update my project to your TTGO version . 
there was alot of problems while compiling with lines :
#86 	//ledcSetup(ledChannel, freq, resolution);
#89 	//ledcAttachPin(tftEnable, ledChannel); that i commented to check back problems. 

i hit on this problem concerning ESP32 library : 

In file included from C:\Users\xxxxxx\Downloads\MagicImageWand-main\MagicImageWand\MagicImageWand.h:37,
                 from C:\Users\xxxxxx\Downloads\MagicImageWand-main\MagicImageWand\MagicImageWand.ino:8:
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:211:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'?
  211 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                ^~
      |                                fs::FS
In file included from c:\Users\xxxxxx\Documents\Arduino\libraries\TFT_eSPI/Processors/TFT_eSPI_ESP32.h:67,
                 from c:\Users\xxxxxx\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:48,
                 from C:\Users\xxxxxx\Downloads\MagicImageWand-main\MagicImageWand\MagicImageWand.ino:7:
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here
   87 | class FS {
      |       ^~
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:211:32: error: 'FS' was not declared in this scope; did you mean 'fs::FS'?
  211 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                ^~
      |                                fs::FS
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\FS\src/FS.h:87:7: note: 'fs::FS' declared here
   87 | class FS {
      |       ^~
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:211:60: error: template argument 1 is invalid
  211 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                                                            ^
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:211:16: error: '<expression error>' in namespace 'std' does not name a type
  211 |   typedef std::function<String(FS &fs, const String &fName)> ETagFunction;
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:212:32: error: 'ETagFunction' has not been declared
  212 |   void enableETag(bool enable, ETagFunction fn = nullptr);
      |                                ^~~~~~~~~~~~
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:229:3: error: 'ETagFunction' does not name a type; did you mean 'OP_Function'?
  229 |   ETagFunction _eTagFunction = nullptr;
      |   ^~~~~~~~~~~~
      |   OP_Function
C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\libraries\WebServer\src/WebServer.h:212:50: error: could not convert 'nullptr' from 'std::nullptr_t' to 'int'
  212 |   void enableETag(bool enable, ETagFunction fn = nullptr);
      |                                                  ^~~~~~~
      |                                                  |
      |                                                  std::nullptr_t

exit status 1

Compilation error: exit status 1

any idea ? 
best regards

NashMerwolf, 2024-09-22:
Figured out that it was ESP32 core 3.0.4 wasnt supported. 
i tried with 2.0.14v and compiled well with lines #89 - 88 - 86 commented// 

Work too with 2.0.17 but not 3.0.0

MartinNohr, 2024-12-09:
Yes, I've been using 2.0.17. I have tried to switch to the 3.x library, but the TFT library doesn't work. I'll figure it out eventually! It could be that the TFT library needs some changes too. I'll post an answer here if and when I get it working, meanwhile just use the older esp library.
Really really sorry about the late response, I've haven't been keeping up with my email.

codenamekraken, 2024-12-25:
same problem how to fix it

codenamekraken, 2024-12-26:
> Figured out that it was ESP32 core 3.0.4 wasnt supported. i tried with 2.0.14v and compiled well with lines #89 - 88 - 86 commented//
> 
> Work too with 2.0.17 but not 3.0.0

Can you please help me buddy

MartinNohr, 2024-12-28:
I can get it to compile but the display no longer works with 3.x.x. Others have also reported this with the TFT_eSPI library. I have not seen a good fix yet. It might be that the original author Bodmer is no longer maintaining that library. I will check in a couple of  months to see if the library gets fixed. If not I have three options.
1. Fix it myself, maybe.
2. Switch to a different display library, might not be easy, I use the sprites from that library.
3. Stay with board version 2.17. Easy for now, but might be a problem in the long run.
I'll post information here when anything changes.