MartinNohr/MagicImageWand #77: Duplicate title listings for SD-CARD images?

pgregg88, 2024-12-08:
Hello. Amazing project! 

I have a large number of images stored on my SDCARD. Each title is duplicated with one title being the actual file and the other being a shadow title that doesn't do anything. As a result, it takes a lot of scrolling to get to the image I need.

Do others experience this? I'm curious if it is something unique to my environment or image publishing pipeline. I create my images on a Mac and save directly to the SDCARD.

Thanks,
Preston

MartinNohr, 2024-12-08:
That's mac creating an attribute (maybe wrong name) file. You could just delete it. Or perhaps I could modify the code to ignore it. Can you send me a sample I can experiment with? I have an old mac but I haven't used it in many years. See if you put the files in a zip file and upload it, then I'll take a look at it. I'm sure I can come up with some way to ignore that extra file. It will still take extra time to read the folder, but that probably isn't too bad.

MartinNohr, 2024-12-08:
I assume that you're talking about the files that start with ._ correct? They hold the Mac file attributes so it can be copied to another Mac. Those files are useless for the MIW. But... I don't know the Mac will let you delete them. It would be easy on a Windows machine to just delete them. I could ignore them in the MIW code, it would slow down the file reading process a little bit, but probably not noticeable. I'll post some code later when I have a few minutes.

MartinNohr, 2024-12-08:
Version 3.28 code will be posted in a few minutes that should fix this problem. I haven't had time to test it yet, so please let me know if the fix works with real MacOS files.

MartinNohr, 2024-12-09:
The code is posted, and the new bin file for the TTGO is here:
https://github.com/MartinNohr/MagicImageWand/tree/main/MagicImageWand/Release
Still, I think the best plan would be to somehow just not put those special MacOS attribute files on the SD card. They don't serve any purpose but do make the directory file larger. Either delete them or figure out how to not create them there in the first place. I would think there must be some way to not create them.

pgregg88, 2024-12-22:
@MartinNohr Sorry for the delayed response. The files starting with ._ is exactly what's going on. I'll get that tested over the holiday. Thanks!

codenamekraken, 2024-12-26:
> @MartinNohr Sorry for the delayed response. The files starting with ._ is exactly what's going on. I'll get that tested over the holiday. Thanks!

Sir can your help me